arduino/Arduino #3661: Add new JTAG library

mrjimenez, 2015-08-11:
Please add this JTAG/XSVF player library.

https://github.com/mrjimenez/JTAG

ffissore, 2015-08-12:
Output is
```
2015/08/12 11:38:18 ... ... tag 1.0.0
2015/08/12 11:38:18 'sentence' and 'url' fields must be defined
```
Please give a second read to https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ and https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

ffissore, 2015-08-12:
Once fixed, please make another tag/release. The job will publish it in at most an hour

closed 2015-08-12 by ffissore

mrjimenez, 2015-08-12:
I'll do the other release. Thanks.

Just one curiosity, "sentence" __was__ defined in library.properties, "url" was not, indeed. I worry that it will fail again, since "sentence" had no apparent trouble. Does the sentence has to end with a stop punctuation?

ffissore, 2015-08-12:
No. Indeed setting url was enough and your lib is now listed. Thanks!