arduino/Arduino #3664: arduino mini pro 3v@8Mhz UART baud rate

antoniosap, 2015-08-13:
Hello, on IDE 1.6.5 linux...
serial.begin(115200) on 8Mhz, receive baud rate dont work. RX not receive at 115200/57600. Now i compile for minipro 5V@16Mhz and set console at 57600. this work.

PaulStoffregen, 2015-08-13:
Of course 115200 doesn't work on an AVR chip running at 8 MHz.  The UART hardware on AVR chips isn't capable of accurate 115200 when running that slow.

antoniosap, 2015-08-14:
the IDE console is set to 115200 and work

matthijskooijman, 2015-08-14:
> Of course 115200 doesn't work on an AVR chip running at 8 MHz.
I have been running 115200 baud on a 8Mhz Pro Mini just fine, connected to a FTDI adapter.

> The UART hardware on AVR chips isn't capable of accurate 115200 when running that slow.
Looking at the  datasheet, at 8Mhz the closest you can get to 115200 is with an -3.5% error. At 16Mhz, you'll have a +2.1% error. So the baud rate is actually significantly worse at 8Mhz, but apparently still good enough to be picked up by an FTDI adapter (Which I suspect can generate the baud rate exactly).

@antoniosap What board are you using exactly? Do you realize that you cannot _change_ the CPU speed by picking a different board in the IDE, but you need to match the selection in the IDE to the hardware you have?

antoniosap, 2015-08-14:
the board is: mini pro 3v@8Mhz, now i set IDE: mini pro 5v@16Mhz at 57600 console and Serial.begin(115200)  different configuration lost RX characters. TX trasmission is ok. (i suspect some bug in IDE profile file)

matthijskooijman, 2015-08-14:
If you have a board running at 8Mhz and you tell the code that it's 16Mhz, I'd expect that all timings are halved, so `Serial.begin(115200)` will really get you 57600, so the setup you described works exactly as expected.

Can you describe a setup that you think is broken?

agdl, 2015-09-23:
@antoniosap I'm closing this for a lack of feedback as suggested by @matthijskooijman. Reopen this if needed

closed 2015-09-23 by agdl