arduino/Arduino #3665: Page on the wiki without content

mrjimenez, 2015-08-13:
Hi,

It seems like this page is messed up, it has important information, I think it would be nice to revert it:

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

Regards,
Marcelo.

PaulStoffregen, 2015-08-13:
Looks like this copy from April 21 is still good

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification/b16d462af0da1b6e0c59b8a0879db64a2694eefb

ffissore, 2015-08-14:
Could not find version "08a93aead0e083fbeecc016aef6ca0797e642b1d"

Gosh, something broke badly

ffissore, 2015-08-14:
Ok, solved the conundrum. I've renamed that page a while ago, removing 2 stale spaces in the title. I have NOT left the original page with a link to the new one. A guy recreated the original page with "hellno" as the only text (useful, very useful). I've recreated that page with the link to the new one

closed 2015-08-14 by ffissore