arduino/Arduino #3666: Pins left as output mode after SPI.end()

nerdralph, 2015-08-14:
SCK, MOSI, and SS are left as output after calling SPI.end().
If SS is in input mode before calling SPI.begin(), it is left as output high.

The comments in SPI.cpp seem to imply the intention was to enable the pullup if the pin wasn't already set to output mode, but then goes on to set it to output (from line 36):
    // if the SS pin is not already configured as an output
    // then set it high (to enable the internal pull-up resistor)
    if(!(*reg & bit)){
      digitalWrite(SS, HIGH);
    }

    // When the SS pin is set as OUTPUT, it can be used as
    // a general purpose output port (it doesn't influence
    // SPI operations).
    pinMode(SS, OUTPUT);

The docs imply that the user code should set the SS pin to output,
https://www.arduino.cc/en/Reference/SPI

If saving and restoring the port state is too complicated, then SPI.end should set the pins to reset state (input low).  I also think controlling SS should be left to the user code, so just delete code in SPI.begin() that manipulates SS.

facchinm, 2015-08-14:
Hi @nerdralph , SPI on AVR has the "strange" feature to configure itself as slave if physical SS pin is an input. So it need to be set as output, no matter if HIGH or LOW (see #2974).

About the other pins not being restored to the original state after `end()` is called, we'd better try to understand what people want to do with that pins after calling `end()`, so maybe it's better to open a thread on the mailing list first.

nerdralph, 2015-08-14:
@facchinim  You're only half right.  SS does NOT have to be configured as output.  One reasonable option is to enable the pullup, which the first part of SPI.begin() does... before setting it to OUTPUT.

From looking at your github account it seems you actually work for Arduino.  I guess Marchese Banzi hires people with no more clue about writing good software than he does, hence the arduino core being the piece of shit it is... 
... and I'm the fool for thinking I could help make it better...

matthijskooijman, 2015-08-14:
@nerdralph, That comment is offensive and totally uncalled for. Feel free to suggest improvements and corrections, but please keep your comments and tone constructive (and no, I'm not on Arduino's payroll, I just care about this community).

As for the SS pin configuration in `SPI.begin()`, I agree that it is confusing and perhaps incorrect. The current code was merged after discussion in #2659 (and possibly a mailing list conversation that is not linked there). I've added a comment to that PR, so let's keep discussion about `begin()` there?

As for restoring the pin modes - can you explain your usecase for calling `SPI.end()`? Typically, sketches only call begin but never end (and I can't think of any obvious reasons to do so anyway, though admittedly I haven't tried very hard).

nerdralph, 2015-08-14:
@matthijskooijman I, like Linus Torvalds, don't give a damn if people's feelings get hurt.  This is computer science, not an emotional support group.
When people say dumb things while assert themselves as knowing what's right, there should be consequences; i.e. calling them out.
Besides the linux kernel maling list, check out eevblog.com for an example of another community that isn't afraid to call, "bullshit", when they see it.

As for why use SPI.end(), you seem to be engaging in a reverse onus logical fallacy; i.e. I should prove why it needs to be fixed rather than just leave it broken.
Nevertheless, here's one sketch that uses SPI.end():
https://github.com/PeterVH/ArduinoISP/blob/master/ArduinoISP/ArduinoISP.ino

facchinm, 2015-08-14:
Well, @nerdralph , I believe you didn't follow the entire conversation (which took several pages of discussions in the mailing list). 

The internal pullups are not enough strong to keep the HIGH state while handling the board, so we decided to set the pin as OUTPUT (thus making some people unhappy) instead than having a formally correct implementation and a lot of subtle bugs (due to how people __use__ these boards). Please not that the previous implementation always set the SS pin as OUTPUT HIGH (you could change it later but the flip was totally unexpected).

I'll be happy to merge a patch which sets SCK and MOSI as input inside `end()` because it is the correct approach

nerdralph, 2015-08-14:
I noticed the bugs and reported them.  The fact that you think an
unconnected SS pin with the pullup enabled will not stay high is further
evidence of your lack of understanding.  Some of my code (multi-byte send
optimizations) was included with the SPI transactions update.
It seems for this I was wasting my time.  If I find any more bugs I wont
bother reporting them and will instead add them to my list of why the
Arduino software sucks for the next time some noob posts how he thinks
Arduino is so great.

matthijskooijman, 2015-08-15:
> As for why use SPI.end(), you seem to be engaging in a reverse onus logical fallacy; i.e. I should prove 
why it needs to be fixed rather than just leave it broken.

I was mostly trying to figure out *how* to best fix this problem, for which usecases help. There are already two alternatives, of which one (storing the original pin modes) has non-negligable cost (in terms of code size, ram usage and engineering effort), so it makes sense to see if the simpler approach (reset to initial state) would be sufficient.

Looking at the ArduinoISP example, AFAIU the intent is that when not programming the target, the MOSI and SCK should be hi-z, so the target can run normally (and act as an SPI master too). For this usecase, simpling reverting the pins to INPUT in `SPI.end()` would be sufficient.

nerdralph, 2015-08-15:
@matthijskooijman What did you think I meant when I said:
"If saving and restoring the port state is too complicated, then SPI.end should set the pins to reset state (input low)."
The Arduino core certainly doesn't need any more bloat, so my suggestion would be:pinMode INPUT.

barsznica, 2018-01-05:
Related to [this](https://github.com/arduino/Arduino/pull/2659), [this](http://forum.arduino.cc/index.php?topic=165497.msg3549880#msg3549880) and/or [this](https://github.com/arduino/Arduino/commit/d101bf51a2c9ebe08be737dd9b5db8b32d64e44a)?