arduino/Arduino #3668: SoftwareSerial on Arduino Micro

a3dbox, 2015-08-14:
I have a strange issue when using software serial on the Arduino Micro.

I can only get SoftwareSerial to work if I also call Serial1.begin().

matthijskooijman, 2015-08-14:
Can you try using the latest 1.6.x version? I'm not sure if anything relevant has changed, but it would be good to know if that version is still broken.

Can you describe in what way you conclude that it does not work? e.g. post a small example sketch and the wiring you apply?

a3dbox, 2015-08-19:
Hi,
Thanks for the reply.  I have updated to 1.6.5 with the same issues.

I have two arduinos wired for I2C comms.  Using the master example on I2C from the android website.  This functions correctly.  

I replace the Wire library on the master with the SoftwareSerial verion and connect to (the same two) different pins on the arduino programmed for the software I2C library.  I do not get any comms on the other arduino.  

I Open Serial1 on the master arduino to print out some debug info and software I2C starts working.

matthijskooijman, 2015-08-19:
Huh? Are you trying to use SoftwareSerial to talk to an I2C device? SoftwareSerial provides an asynchronous serial connection, not I2C?

a3dbox, 2015-08-19:
LOL or really sorry!  That is confusing, I am working on I2C this morning and am mixing things up.  let me write that again...  Please ignore my previous post!

I have two arduinos wired together for serial comms.  If I write serial from one arduino to the other using Serial.println() I can receive the message on the other arduino.

I replace Serial with the SoftwareSerial version and re configure my wiring to different pins on the arduino based on the software serial pins. I do not get any comms on the other arduino.

I Open Serial1 on the sending arduino (Serial1.begin(<speed>)), but do not actually use it to send anything and software serial starts working.

matthijskooijman, 2015-08-19:
That's certainly weird. What Arduinos are you using? Can you post your code?

a3dbox, 2015-08-19:
Using Arduino Mircos.

Would an example project suffice?  I cant share the project.

NicoHood, 2015-08-19:
The more abstract it is (easier to reproduce) the better it is. Yes.

NicoHood, 2015-09-02:
Any example?

a3dbox, 2015-09-02:
Nico,

 

Really sorry, work has pulled me onto a different project, I still have this issue but don’t have any time at the moment to look at it.

As soon as I am back on the project I will post an example.

agdl, 2015-09-24:
@a3dbox any news?

agdl, 2015-11-26:
I'm closing this for lack of feedback. Please reopen it if needed

closed 2015-11-26 by agdl