arduino/Arduino #3669: Bug with Serial.println("!!!!!!!!!!!!!! FACTORY RESET !!!!!!!!!!!!!!!");

johnlenfr, 2015-08-14:
Hi, on Arduino IDE 1.6.5, when I use Serial.println("!!!!!!!!!!!!!! FACTORY RESET !!!!!!!!!!!!!!!"); in the sketch, I can't upload the sketch in my arduino board MEGA2560.
I have to unplug the board and I have the error:
"Arduino : 1.6.5 (Windows 7), Carte : "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Le croquis utilise 2 992 octets (1%) de l'espace de stockage de programmes. Le maximum est de 253 952 octets.

Les variables globales utilisent 244 octets (2%) de mémoire dynamique, ce qui laisse 7 948 octets pour les variables locales. Le maximum est de 8 192 octets.

avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: stk500v2_ReceiveMessage(): timeout

avrdude: ser_send(): write error: sorry no info avail

Problème de téléversement vers la carte. Voir http://www.arduino.cc/en/Guide/Troubleshooting#upload pour suggestions.

  Ce rapport contiendrait plus d'informations si l'option
  "Montrer les informations de sortie pendant la compilation"
  était activée dans Fichier > Préférences.
"
Have you got the same problem? Is it due to the '!' caracter?

Thanks for reply.
John

I tested with the Blink sketch:

/*
  Blink
  Turns on an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the Uno and
  Leonardo, it is attached to digital pin 13. If you're unsure what
  pin the on-board LED is connected to on your Arduino model, check
  the documentation at http://www.arduino.cc

  This example code is in the public domain.

  modified 8 May 2014
  by Scott Fitzgerald
 */

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);
  Serial.begin(115200);
}

// the loop function runs over and over again forever
void loop() {

  Serial.println("!!!!!!!!!!!!!! FACTORY RESET !!!!!!!!!!!!!!!");
 
  
  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
}

facchinm, 2015-08-14:
This is a known bug, that was fixed in newer bootloader version. See also #1594 and #392.

closed 2015-08-14 by facchinm

johnlenfr, 2015-08-14:
Ok, sorry for the inconvenience.

Will update my bootloader