arduino/Arduino #3675: build.extra_flags -U undef option not working

NicoHood, 2015-08-14:
I want to deactivate the USB core of Arduino. Therefore USBCON needs to be undefined. Passing options with -D works, -U not.

2nd issue related to this: the USB-Core should not rely on such a USBCON definition, there should be a wrapper

cc @facchinm

matthijskooijman, 2015-08-14:
@NicoHood AFAIU, -U only undefines things previously defined with -D on the same commandline, it does not prevent #defining them in the source. To support what you need, you would need an #ifnded NO_USB (or something better named) in the source.

NicoHood, 2015-08-15:
Okay so I changed the issue.

We could:
* integrate a #ifndef NO_USB_CORE as you said
* or we could make the whole USB-Core a library which can be included.

2nd option would save us any compiler flags and new boards.txt submenus.
cc @facchinm

matthijskooijman, 2015-08-15:
Problem with the second option is that any existing sketches will break, since they won't have the right include...

NicoHood, 2015-08-15:
See the PR I referenced. This would solve it properly.

The only question is if we should add a workaround or update the bootloader.