arduino/Arduino #3677: Need to include at least one root directory header file

lchristina26, 2015-08-14:
I have found that my library needs to include at least one header file in the main directory. For example, when I go to ```Sketch > Include Library > example_lib```  it will pull in ```libraries/example_lib/header1.h``` automatically into my sketch. If example_lib does not have any header files in the main directory, then I cannot include any other header files. Say I have ```example_lib/headers/header2.h``` and try to include that in my Arduino sketch:

    #include <headers/header2.h>

 it reports an error:
 
    fatal error: headers/header2.h: No such file or directory

But if I put header1.h back in the main directory, and do ```Sketch > Include Library > example_lib``` again it will automatically pull in the main directory headers, in my case just ```#include header1.h```, I can include ```headers/header2.h``` and any other header files in subdirectories of my examle_lib.

Also, if I remove a file and add it again, I have to go to ```Sketch > Include Library > Library Manager ``` and then just close that screen after it opens or it will not catch my changes. 

Shouldn't it find the header files in subdirectories automatically? Why does having main directory header files allow me to have other subdirectory files?

matthijskooijman, 2015-08-15:
The procedure is that the IDE checks the includes in the .ino file, and finds a library for every include. Then, during compilation, the found libraries are put in the compiler's include path. The first step only looks at files in the top level directory (or `src/` directory for libraries using the alternative layout). The second step does look at subdirectories, which creates this discrepancy.

I'm not sure if this is fully intentional, but I can imagine that including library subdirectories in the search could increase the chance of collision between two libraries. Als note that, unless you use the `src/` layout, any source files (.c / .cpp) in subdirectories will not be compiled, even though header files (.h) can be included (by specifying the name of the subdirectory explicitly).

> Also, if I remove a file and add it again, I have to go to Sketch > Include Library > Library Manager and then just close that screen after it opens or it will not catch my changes. 

I think this is because the list of available libraries is not rescanned on every compile, just on startup and some moments (like opening or closing the lib manager).

I hope this clarifies why things work as they do. I'm not sure if there is anything that needs to be changed, though.

closed 2015-08-17 by ffissore

lchristina26, 2015-08-17:
Thanks, that does explain things. However, there are many libraries that exist where source files are included in subdirectories, not necessarily in /src. It would be convenient if Arduino had a way to find these .c/.cpp files rather than the user having to change the structure of the library. 

Thanks again for your clarification.

matthijskooijman, 2015-08-17:
> However, there are many libraries that exist where source files are included in subdirectories, not necessarily in /src. 

Do you mean Arduino libraries here? Or just libraries in general? Right now, the only way to make the Arduino IDE compile files in subdirectories, is to put them under "src/".