arduino/Arduino #3682: [DOC] https://www.arduino.cc/en/Reference/DueExtendedSPI links to ¿obsolete? API

q2dg, 2015-08-15:
Specifically it says: "See the individual reference pages for setClockDivider(), setDataMode(), transfer(), setBitOrder() for proper syntax when using the extended methods. " 
But in all these individual reference pages (except for transfer()), the first sencence is: "This function should not be used in new projects. Use SPISettings with SPI.beginTransaction() to configure SPI parameters." 
I think https://www.arduino.cc/en/Reference/DueExtendedSPI should be rewritten with new API.

agdl, 2015-09-04:
@q2dg i put in bold in all the pages that they are not suggested because are deprecated.

q2dg, 2015-09-04:
Ok

q2dg, 2015-09-04:
Well, in https://www.arduino.cc/en/Reference/SPI there is a paragraph titled "Extended SPI functionality for the Due" that maybe should be removed or marked as obsolete...

q2dg, 2015-09-04:
And in https://www.arduino.cc/en/Reference/SPITransfer , what happen with the text below "Extended method for Arduino Due" section??

agdl, 2015-09-07:
@q2dg removed in the SPI reference. I think should be removed in transfer as well and add informations as I suggested in https://github.com/arduino/Arduino/issues/3685

PaulStoffregen, 2015-09-07:
This page is unclear.  It covers multiple APIs.

https://www.arduino.cc/en/Reference/DueExtendedSPI

SPI.setClockDivider(), SPI.setBitOrder() and SPI.setDataMode() are deprecated.

The Due extended transfer is *not* deprecated.

To properly update this page, it must be rewritten by a person with good understanding of the SPI library API.  The example code needs to remove the deprecated functions and add the SPISettings and SPI.beginTransaction() and SPI.endTransaction() functions.

It can not be "fixed" by simply adding a quick but unclear notice.

agdl, 2015-09-07:
@PaulStoffregen I know you are very good with SPI and I'm not, if you want you can tell me what to add to the SPI pages to have a clear documentation and I will add these informations to make people happy in the better way. Unfortunately at the moment we don't have a contributing system that is differ from: tell what you want to add and I will do it.
I'm sorry

PaulStoffregen, 2015-09-07:
I have not used the Due extended API.  This is really a question for @cmaglie

PaulStoffregen, 2015-09-07:
Do you have an Arduino Due board?  Maybe you could connect a SPI chip and attempt to use it?

cmaglie, 2015-09-08:
Hi Paul!

I've internally acknowledged to @agdl that the Extended API could be marked "deprecated".

Let me recap the scope of the various SPI APIs, for @agdl and others benefit.

The **"classic" SPI API** has:

`void begin()`
`byte transfer(data)`
`void setBitOrder(bitorder)`
`void setDataMode(datamode)`
`void setClockDivider(clockdiv)`
`void end()`

This is the oldest and simpler API, but has some drawbacks:

- you must manually handle the CS pin using `pinMode` and `digitalWrite`
- the SPI settings are unique for the SPI bus: if you have multiple devices connected you might need to change settings every time you talk to each device
- the `setClockDivider` is not portable: the actual SPI clock speed depends on the main CPU clock.
- there is no protection if you use SPI inside interrupts ISR

With the introduction of SAM3X and the Arduino Due we added the "Extended SPI" API to exploit the hardware capabilities of the SAM3X to automatically handle the CS pin assertion and automatically set SPI parameters for each device.
The **Extended SPI API** adds the following methods to the classic API:

`void begin(pin)`
`byte transfer(pin, data, SPI_LAST/SPI_CONTINUE)`
`void setBitOrder(pin, bitorder)`
`void setDataMode(pin, datamode)`
`void setClockDivider(pin, clockdiv)`
`void end(pin)`

where `pin` is the CS pin for each device. Every pin has his own settings and the `transfer(pin, ...` function assert and de-assert CS (and has an additional parameter LAST/CONTINUE to tell the CPU if the CS pin should be kept asserted when transferring more than one byte).

BTW, even if this may sounds as a cool feature of the SAM3X chip, in the real world it turns out that it has some nasty drawbacks, in particular:

- you can use only some specific pins for CS (on the Due board are pins 4, 10, 52 and another pin that is not broken out).
- in the market there are some SPI devices that needs a non-standard CS pin handling (some popular RF modules for example). If you want to use such devices you're basically screwed up and you must revert to "classic" (not-extended) SPI.

Knowing about those drawbacks a couple of years ago would probably let us decide to not introduce the Extended API at all. Whatever, what is done is done: now we have it and, even with these limitations, it works.

Finally, following your proposal in devlist, we introduced the **Transaction SPI API**: the main reason behind this new API is to add protection for devices that uses SPI inside ISR (every day more and more common), but also to handle different SPI setting for each device efficiently even if the CPU doesn't have specific hardware capabilities to do that.

The following methods were added:
`usingInterrupt(interruptNumber)`
`settings = SPISetting(clock, bitOrder, dataMode)`
`beginTransaction(settings)`
`endTransaction()`

With this API:
- multiple SPI settings are now elegantly solved with SPISettings
- the CS pin handling is left to the user (so devices with non-standard CS behaviour can be used)
- the `usingInterrupt` method allow to mask problematic ISR when we are inside a `beginTransaction`/`endTransaction` block.
- also the non-portable `setClockDiv` has been replaced with the SPISettings `clock` parameter (where you set the effective maximum clock allowed by the device in Hz)

So, to conclude, IMHO the documentation should mark as deprecated both the "old" and the "extended" API, because both are in effect superseded by Transactions.

PaulStoffregen, 2015-09-08:
Oh, ok, I did not realize the Due extended API is now deprecated.  I guess I don't know much about Arduino's internet dialogue these days....

agdl, 2015-09-10:
Ok so let's make a little recap. I removed the extended reference like said in #3685  in the transfer and wrote here

https://www.arduino.cc/en/Reference/DueExtendedSPI

that these methods should not be used in new projects and that is better to use SPI settings. What can we do to improve the documentation?

We need to add:
SPI.transfer(void *buf, size_t count)
SPI.transfer16(unsigned int data)

Nothing else?

q2dg, 2015-09-10:
Maybe you should grep "Extended  SPI" or "Extended method" through all SPI-related pages to remove any indication of this Due api...For instance, https://www.arduino.cc/en/Reference/SPIBegin has a pharagraph about it without any notice about deprecation.

agdl, 2015-09-11:
Maybe it is better to totally remove the reference in any page? If someone needs the exetended methods he directly search for "arduino SPI extended methods " doesn't he?

q2dg, 2015-09-11:
I think so

agdl, 2015-09-11:
please take a look at : https://www.arduino.cc/en/Reference.SPITransfer

q2dg, 2015-09-11:
*receivedVal = SPI.transfer16(val16)* shouldn't be *receivedVal16 = SPI.transfer16(val16)* ??

It isn't clear if *SPI.transfer(buffer, size)* returns something and if so, what is it (receivedVal, receivedVal16, a buffer with the same size) ??

agdl, 2015-09-11:
It is written this:
In case of buffer transfers the received data is stored in the buffer in-place (the old data is replaced with the data received).

Practically the buffer is overwritten with the received data

q2dg, 2015-09-11:
Ok, sorry

agdl, 2015-09-14:
I removed the Extended methods in all the pages where it was cited, but left it in the See Also. What do you think about this solution?

q2dg, 2015-09-14:
For me it's ok, so I'll close the issue.

But I insist: in https://www.arduino.cc/en/Reference.SPITransfer, 
*receivedVal = SPI.transfer16(val16)* shouldn't be *receivedVal16 = SPI.transfer16(val16)* ??

closed 2015-09-14 by q2dg

agdl, 2015-09-14:
@q2dg yes done