arduino/Arduino #3683: [DOC] Inconsistencies in SPI Examples

q2dg, 2015-08-15:
1.- https://www.arduino.cc/en/Tutorial/SPIEEPROM is too low level...in fact, doesn't use SPI library but AVR registers. It could remains as it is but a note should be placed indicating this tutorial is only valid for AVR-based boards (and I don't know if for all of them).

2.-https://www.arduino.cc/en/Tutorial/BarometricPressureSensor talks about a retired product and moreover (but that's is a personal opinion) the code is too complicated. Anyway, the code should updated with new SPI API.

3.-https://www.arduino.cc/en/Tutorial/SPIDigitalPot 's code only needs to be updated with new SPI API

Finally, https://www.arduino.cc/en/Tutorial/SPITransaction seems fine, but, as it is said in https://github.com/arduino/Arduino/issues/2654, something must be done with the management of repo docs in relation with web docs

agdl, 2015-09-02:
@q2dg take a look at this. If they are ok for you I will make a PR
https://github.com/agdl/Arduino/commit/29c9fb4ea64213c9eea1b1c1a31d8f6348d24ecf

For what concern point 1 you are right. What can be the best way to place this note?
Maybe something like: 
This example shows a low level implementation of an SPI communication on an Arduino UNO. More user friendly examples can be found here (link to the others).

Let me know!

q2dg, 2015-09-02:
I like the note you suggest, and the changes seems ok. So I close the issue, ok?

Although I continue thinking BarometricPressureSensor is not the best example to start, but never mind.

closed 2015-09-02 by q2dg