arduino/Arduino #3685: [DOC] SPI.transfer(void *buf, size_t count) and SPI.transfer16(unsigned int data) aren't documented in web

q2dg, 2015-08-16:
But they exist at https://github.com/arduino/Arduino/blob/bd8f7932e64fb3a41e3d934437d5eabd8707dcc9/hardware/arduino/avr/libraries/SPI/SPI.h

q2dg, 2015-08-16:
Well, I see in https://github.com/arduino/Arduino/blob/master/hardware/arduino/sam/libraries/SPI/SPI.h and in https://github.com/arduino/ArduinoCore-samd/blob/master/libraries/SPI/SPI.h that Due and Zero (respectively) only has SPI.transfer(void *buf, size_t count) implemented, but they hasn't SPI.transfer16(unsigned int data). Is it intentional?

Anyway, at least the transfer() variant should be advertised, don't you think? 
Thanks.

agdl, 2015-09-04:
@q2dg actually with the introduction of the transactions, the transfer function is deprecated and hold only for retro-compatibility reasons. So maybe it's better to not add documentation about this.

q2dg, 2015-09-04:
What? The transfer function is the only way to transfer bytes inside a SPI communication, either inside a transaction or not!!!!!!.  Are you saying these two alternatives (SPI.transfer(void *buf, size_t count) and SPI.transfer16(unsigned int data) are deprecated? Because SPI.transfer(byte data) isn't!!!!

PaulStoffregen, 2015-09-06:
The deprecated functions are SPI.setClockDivider(), SPI.setBitOrder() and SPI.setDataMode().

SPI.transfer() is not deprecated.

agdl, 2015-09-07:
@q2dg @PaulStoffregen I know, you are right, I made a mistake. I will add these functions.

agdl, 2015-09-07:
What do you think about documenting them in the transfer page and delete (in the transfer page) the informations about the extended methods for due?

agdl, 2015-09-07:
cc @cmaglie

cmaglie, 2015-09-08:
For me it's fine to remove "Extended SPI" methods on the `transfer` page.

Probably the Extended API is obsoleted by transactions, I've made a more in-depth analysis in #3682. Let's move the discussion there and close this one if you're done.

agdl, 2015-09-10:
as suggested by @cmaglie i removed the reference to extended methods and closing this!

closed 2015-09-10 by agdl