arduino/Arduino #3686: MQ Sensors for Arduino.

Sanpasqual123, 2015-08-16:
AQ-3 is not reactive to alcohols or hydrocarbons, being selective for hydrogen.  See manufacturer data sheet: http://www.arihav.com/Datasheet/SP-AQ3.pdf

matthijskooijman, 2015-08-16:
I'm sorry, but what do you want to achieve with this issue exactly? If you want a library to be included in the library manager, you'll have to be more explicit. If you want someone to write a library to talk to a specific sensor, this is not the right place to ask (the arduino forums on arduino.cc would be more appropriate).

Sanpasqual123, 2015-08-16:
I was pointing to an error on the extant listing.

matthijskooijman, 2015-08-16:
And with extant you mean existent? :-)

A listing of a library? Which library? If the error is in the description for a library, you should probably report the error to the library maintainer, since Arduino just displays whatever description is stored inside the library.

ffissore, 2015-08-17:
Closing as invalid

closed 2015-08-17 by ffissore