arduino/Arduino #3688: Update of Library Manager seems to be broken

plinioseniore, 2015-08-16:
Hi,

just released a new tag for souliss (https://github.com/souliss/souliss/releases) that hasn't been catch by the script that updates the Library Manager (that has far as I know runs at 1h interval).

Please verify.

Thanks,
Dario.

plinioseniore, 2015-08-16:
Haven't waited enough, now is working.

plinioseniore, 2015-08-16:
Having a detail look, the JSON was updated correctly and looking at it through the browser gives also the last tag, but this wasn't recognized in Arduino IDE.

The only way to have the last tag available in the Arduino IDE is to delete the json file from the %APPDATA% folder and lets the Library Manager download it again.

I'm running the IDE on Windows 10.

ffissore, 2015-08-17:
Works for me. I guess your IDE was fetching a cached json file. It may be due to using an not yet synchronized download server or some stuff running on your network (proxies, "web accelerators")

closed 2015-08-17 by ffissore

plinioseniore, 2015-08-17:
It wasn't from the browser I was able to get the JSON file and see the
updated one.

The IDE wasn't able to replace the file in the %appdata%.

I will try again and let you know.