arduino/Arduino #3693: Serial port - spaces in name

beswickat, 2015-08-17:
On my Mac, the serial port appears as /"dev/cu.wch ch341 USB=>RS232 1440", including spaces which causes an error message like "avrdude: ser_open(): can't open device "/dev/cu.wch": No such file or directory"

Can't rename device to avoid the spaces.

Work around: 
sudo ln cu.wch\ ch341\ USB\=\>RS232\ 1440 cu.wchrint

But this command has to be issued every time the USB is connected.

ffissore, 2015-08-18:
Also see #2244 which has been closed as wontfix

q2dg, 2015-08-18:
It's not an official board, so there shouldn't be official support. I'd close this issue.

matthijskooijman, 2015-08-18:
Is there a reason to not fix this anyway? Seems like it's a matter of [changing platform.txt](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/platform.txt#L94) to add double quotes around the -P option?

@beswickat, could you try this perhaps? Find the `arduino/avr/platform.txt` file (somewhere within the Arduino app), find the line I linked above and replace it with:

    tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} "-P{serial.port}" -b{upload.speed} -D "-Uflash:w:{build.path}/sketch/{build.project_name}.hex:i"

If that works, do you know how to submit a pullrequest by any chance?

beswickat, 2015-08-18:
Hi
I just figured out a permanent fix for Mac and UNIX

cd /dev
sudo ln -s cu.wch\ ch341\ USB\=\>RS232\ 1440 cu.wch

This creates a shortcut to the name with spaces in it. The shortcut is not deleted when the IDE quits, and next time the IDE starts, the shortcut is there, waiting to point to the device whose name has spaces in it.

No idea about Windoze

cheers
Alan

beswickat, 2015-08-18:
Hi Matthijs
I have just discovered a work around :-)

On the Mac and other UNIX beasts, make a SYMBOLIC link
cd /dev
sudo ln -s cu.wch\ ch341\ USB\=\>RS232\ 1440 cu.wch

Now, /dev/cu.wch is a ‘shortcut’ to /dev/cu.wch\ ch341\ USB\=\>RS232\ 1440, and /dev/cu.wch can exist without /dev/cu.wch\ ch341\ USB\=\>RS232\ 1440 existing.

The serial device with the long name disappears when you shut down the Arduino, but the symbolic remains as a dangling link, normally to be avoided as it is a shortcut to a non-existing entity. However, when the Arduino IDE starts again, the serial device with the spaces in the name is recreated and the symbolic link now points to it again.

This is the first time in 30 years of UNIX programming that I have intentionally used a dangling link, but it works. :-)

On a Windoze box, a short cut to the appropriate file might work.

Also, I don’t know what a pull request is, sorry.

cheers
Alan (Brisbane, Australia)

smartperson, 2017-06-15:
@matthijskooijman @beswickat I can confirm that this issue is resolved by editing platform.txt at `Arduino.app/Contents/Java/hardware/arduino/avr/programmers.txt`. I'll submit a pull request which might be rejected, as I believe new versions of the CH340 driver fix this issue. Perhaps Arduino official doesn't care about this issue, but it's good form to escape paths like these. The other paths on that line are double-quote wrapped, after all…

I had quite an experience tracking this problem down. I posted on [my blog](http://varunmehta.com/technology/2017/06/14/mbot-with-macos-lion.html) about its impact and the solution @matthijskooijman mentioned.

matthijskooijman, 2017-06-15:
@smartperson, that sounds like a familiar kind of ride: working your way through an issue and only when you completely diagnose it and know how to solve it, you have sufficient info to find an existing bug report for the issue :-)

In any case, thanks for testing the fix and the PR, I think there's no problem in merging it.

beswickat, 2017-06-15:
Hi
thanks for you efforts, 
cheers
Alan (Australia)

PR #6383 (2017-07-25): Fixed serial port argument that may include spaces
Other paths in avrdude.upload.pattern are wrapped in double quotes, and this -P{serial.port} causes issues with some platforms. This allows serial port devices which include spaces and other characters. Without this fix a /dev/tty* or /dev/cu* device that includes a space in its name gets truncated when passed to avrdude. Error messages returned from avrdude are cryptic, and workarounds (symlinks) are prone to failure.

Fixes #3693
commit 0fc235da (2017-07-25): Fixed Serial Port Argument That May Include Spaces

Other paths in avrdude.upload.pattern are wrapped in double quotes, and
this -P{serial.port} causes issues with some platforms. This allows
serial port devices which include spaces and other characters. Without
this fix a /dev/tty* or /dev/cu* device that includes a space in its
name gets truncated when passed to avrdude. Error messages returned
from avrdude are cryptic, and workarounds (symlinks) are prone to
failure.

Fixes #3693
closed 2017-07-25 by cmaglie