arduino/Arduino #3695: DIY Gamer Kit libraries for library manager

mikevanis, 2015-08-18:
Hi! 

We'd like to include the [DIY Gamer Kit](https://www.techwillsaveus.com/shop/diy-kits/diy-gamer-kit-arduino/) library in the library manager. 

Here is a link to the release: https://github.com/techwillsaveus/Gamer/releases/tag/v2.0

Additionally, Gamer has another library for interfacing with multiplayer (IR serial) functions. 

Here is a link to that release: https://github.com/techwillsaveus/GamerIR/releases/tag/v1.0

Many thanks! Looking forward to our library appearing gloriously in the brand new (awesome) library manager!!

TWSU

ffissore, 2015-08-18:
Added. Thanks!

mikevanis, 2015-08-19:
Hi @ffissore, thanks for adding our libraries! GamerIR works perfectly. However, Gamer throws an error when we try to install it with the library manager. This is the error we get: 

`CRC doesn't match. File is corrupted`

ffissore, 2015-08-19:
Did you force pushed a new tag?

mikevanis, 2015-08-19:
I don't think we did. The latest commit is tagged as v2.0, and was created with GitHub's release feature. Any suggestions / wisdom?

ffissore, 2015-08-19:
Probably some change from github. I'll regenerate the hash

mikevanis, 2015-08-19:
Great! Feel free to give me a ping if it starts working. We're trying from our end too!

ffissore, 2015-08-19:
It's working now. You'll have to manually delete previously downloaded .zip file (.arduino15/staging/libraries)

closed 2015-08-19 by ffissore

mikevanis, 2015-08-19:
Perfect, thanks Federico! Thumbs up :+1: