arduino/Arduino #3696: SD card example fail

maxcpr, 2015-08-18:
a little bit changing example from IDE (SD listfiles), file listing moved to "loop".
Code stops listing microSD card after 17-20 loops (arduino nano).

/*
  Listfiles
 
 This example shows how print out the files in a 
 directory on a SD card 
 	
 The circuit:
 * SD card attached to SPI bus as follows:
 ** MOSI - pin 11
 ** MISO - pin 12
 ** CLK - pin 13
 ** CS - pin 4

 created   Nov 2010
 by David A. Mellis
 modified 9 Apr 2012
 by Tom Igoe
 modified 2 Feb 2014
 by Scott Fitzgerald
 
 This example code is in the public domain.

 */
#include <SPI.h>
#include <SD.h>

File root;

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

  Serial.print("Initializing SD card...");
  digitalWrite(10, 1);   // turn the LED on (HIGH is the voltage level)
  digitalWrite(4, 0);   // turn the LED on (HIGH is the voltage level)
  delay(2000); 
  while (!SD.begin(4)) {
    Serial.println("initialization failed!");
    delay(2000); 
//    return;
  }
  Serial.println("initialization done.");
}

void loop(){
  Serial.println("starting!");
  root = SD.open("/");
  printDirectory(root, 0);
  Serial.println("done!");
  delay(2000);
  pinMode(10, OUTPUT);
  pinMode(4, OUTPUT);
  digitalWrite(10, 1);   // turn the LED on (HIGH is the voltage level)
  digitalWrite(4, 0);   // turn the LED on (HIGH is the voltage level)

  
}

void printDirectory(File dir, int numTabs) {
   while(true) {
     
     File entry =  dir.openNextFile();
     if (! entry) {
       // no more files
       break;
     }
     for (uint8_t i=0; i<numTabs; i++) {
       Serial.print('\t');
     }
     Serial.print(entry.name());
     if (entry.isDirectory()) {
       Serial.println("/");
       printDirectory(entry, numTabs+1);
     } else {
       // files have sizes, directories do not
       Serial.print("\t\t");
       Serial.println(entry.size(), DEC);
     }
     entry.close();
   }
}

VogonJeltz, 2015-09-12:
I made it similar, providing a dedicated SDinit() function and aborting after 20 sec timeout:
~~~
#define fileIO_OK            +1
#define fileIO_ERR_CREATE    -1
#define fileIO_ERR_OPEN      -2
#define fileIO_ERR_REMOVE    -3
#define fileIO_ERR_WRITE     -4
#define fileIO_ERR_READ      -5
#define fileIO_ERR_IMPLAUS   -6
#define fileIO_ERR_NAME      -8
#define fileIO_ERR_SDCARD   -16

//=====================================================================================
// SD init
//=====================================================================================

int16_t SDinit() {
   char sbuf[50];
   uint32_t  tstamp;
   int16_t   ior=0;

   pinMode(13, OUTPUT);
   digitalWrite(13, LOW);
   tstamp = clock();
   ior=SD.begin(sd_cs);  // true on sucess; else false
   while( !ior) {      
      sprintf(sbuf, "#: ...SD not found... ");  Serial.println(sbuf);
      delay(1000);   
      ior=SD.begin(sd_cs);
      if (clock()-tstamp>20000) {Serial.println("#: ...break!"); break; }
   } 
  if(!ior) return fileIO_ERR_SDCARD;     // SD ioresult == 0
  digitalWrite(13, HIGH);
  return fileIO_OK ;            // SD init OK  == 1
} 
~~~

jpk-0320, 2016-04-03:
Ok folks new to this. So, I bought a couple of SD card readers, just ran a test on one I complied the ReadWrite program the first time no problem, tried it again and it came back with "GetFileAttributesEx C:\Arduino\libraries\SD\src\File.cpp C:\Arduino\libraries\SD\src/SD.h: The filename, directory name, or volume label syntax is incorrect.
Error compiling." So to test the problem I just compiled a program with SD.h, SPI.h and setup, and loop with no CODE. I get the same error after compiling it the second time. ????

facchinm, 2016-04-05:
Hi @jpk-0320 ,
your problem is different (and related with arduino-builder).
Can you post your PC specifications, IDE version and complete log message (after enabling verbose compilation on the IDE preferences)?

jpk-0320, 2016-04-05:
dell optiplex 790

windows 7 home premium 

intel(r) core(tm) i5-2400 cpu @3.1ghz
8.00 gb 7.88 gb usable
64 bit operating system

================================================================
WORKED

C:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Arduino\hardware" -hardware "C:\Users\(c) jpk\AppData\Local\Arduino15\packages" -tools "C:\Arduino\tools-builder" -tools "C:\Arduino\hardware\tools\avr" -tools "C:\Users\(c) jpk\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Arduino\libraries" -libraries "C:\Users\(c) jpk\Documents\Arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10607 -build-path "C:\Users\(C)JPK~1\AppData\Local\Temp\build7e5bbf331067e824a694bdbf33745ad9.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "C:\Users\(c) jpk\Documents\Arduino\FirstProblem\FirstProblem.ino"
C:\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Arduino\hardware" -hardware "C:\Users\(c) jpk\AppData\Local\Arduino15\packages" -tools "C:\Arduino\tools-builder" -tools "C:\Arduino\hardware\tools\avr" -tools "C:\Users\(c) jpk\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Arduino\libraries" -libraries "C:\Users\(c) jpk\Documents\Arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10607 -build-path "C:\Users\(C)JPK~1\AppData\Local\Temp\build7e5bbf331067e824a694bdbf33745ad9.tmp" -warnings=default -prefs=build.warn_data_percentage=75 -verbose "C:\Users\(c) jpk\Documents\Arduino\FirstProblem\FirstProblem.ino"
"C:\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\standard" "C:\Users\(C)JPK~1\AppData\Local\Temp\build7e5bbf331067e824a694bdbf33745ad9.tmp\sketch\FirstProblem.ino.cpp" -o "nul"
"C:\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\standard" "-IC:\Arduino\libraries\SD\src" "C:\Users\(C)JPK~1\AppData\Local\Temp\build7e5bbf331067e824a694bdbf33745ad9.tmp\sketch\FirstProblem.ino.cpp" -o "nul"
"C:\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\standard" "-IC:\Arduino\libraries\SD\src" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\libraries\SPI\src" "C:\Users\(C)JPK~1\AppData\Local\Temp\build7e5bbf331067e824a694bdbf33745ad9.tmp\sketch\FirstProblem.ino.cpp" -o "nul"
"C:\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\standard" "-IC:\Arduino\libraries\SD\src" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\libraries\SPI\src" "C:\Arduino\libraries\SD\src\File.cpp" -o "nul"
"C:\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\standard" "-IC:\Arduino\libraries\SD\src" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\libraries\SPI\src" "C:\Arduino\libraries\SD\src\SD.cpp" -o "nul"
"C:\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\cores\arduino" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\variants\standard" "-IC:\Arduino\libraries\SD\src" "-IC:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\libraries\SPI\src" "C:\Arduino\libraries\SD\src\utility\Sd2Card.cpp" -o "nul"
"C:\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_UNO -DARDU [... cut ...] cpp" -o "C:\Users\(C)JPK~1\AppData\Local\Temp\build7e5bbf331067e824a694bdbf33745ad9.tmp\sketch\FirstProblem.ino.cpp.o"
Using library SD at version 1.0.6 in folder: C:\Arduino\libraries\SD 
Using library SPI at version 1.0 in folder: C:\Users\(c) jpk\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.10\libraries\SPI 
GetFileAttributesEx C:\Arduino\libraries\SD\src\File.cpp C:\Arduino\libraries\SD\src/SD.h: The filename, directory name, or volume label syntax is incorrect.
Error compiling.

jpk-0320, 2016-04-05:
Sorry folks about the above mess

jpk-0320, 2016-04-05:
1.6.7 is IDE version I am using

jpk-0320, 2016-04-05:
the code
#include <SD.h>
#include <SPI.h>
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

jpk-0320, 2016-04-05:
i put in the SD.h header file and the SPI.h header file it does not show up here ????

agdl, 2016-07-12:
This issue was moved to arduino-libraries/SD#4

closed 2016-07-12 by agdl