arduino/Arduino #3700: Documentation for attachInterrupt is misleading

nickgammon, 2015-08-19:
The documentation for attachInterrupt here:

https://www.arduino.cc/en/Reference/attachInterrupt

says: 

> Note
>
> Inside the attached function, delay() won't work and the value returned by millis() will not increment.

This is correct. Further down the page however it says:

>  as delay() and millis() both rely on interrupts, they will not work while an ISR is running

Yes, millis() will work. It will return the **current value**. It is misleading to say it will not work. It is quite acceptable for an ISR to record the current value returned by millis() as being the time something happened.

This "will not work" phrase is confusing users into thinking they cannot record the moment an ISR occurred, using `millis()`.

matthijskooijman, 2015-08-19:
Agreed, though if an ISR runs for > 1ms, `millis()` will stop being incremented at some point, it can even jump down again IIRC, so use of `millis()` isn't entirely without limitations.

nickgammon, 2015-08-19:
Certainly it is not without limitations, and that is covered by saying: 

> millis() will not increment

It is incorrect and misleading to say it won't "work". How can a function that returns a number (and does nothing else) not "work"? It might return the wrong number, certainly. However the fact is that, during the operation of the ISR, it will return the same number - that number being the millis() counter at the time the ISR was entered. Hence it is correct to say it will not increment, but incorrect to say it won't "work".

nickgammon, 2015-08-19:
> Agreed, though if an ISR runs for > 1ms, millis() will stop being incremented at some point, it can even jump down again IIRC, so use of millis() isn't entirely without limitations.

It won't be incremented at any point during the ISR.

`micros()` may jump down during a long ISR because it consists of two parts - the saved counter and the contents of the hardware timer register. If you wait too long the timer register will wrap around twice and then micros() could go down.

matthijskooijman, 2015-08-19:
I think we are in agreement - the current text is wrong. I just meant that `millis()` doesn't work entirely as normal so a not should be present, but there is already a note there I see now.

> micros() may jump down during a long ISR because it consists of two parts - the saved counter and the contents of the hardware timer register. If you wait too long the timer register will wrap around twice and then micros() could go down.

Ah, indeed, I was thinking about this behaviour exactly. Of course that's `micros()`, not `millis()`.

How about replacing this:

> as delay() and millis() both rely on interrupts, they will not work while an ISR is running. delayMicroseconds(), which does not rely on interrupts, will work as expected. 

with this:

> `millis()` relies on interrupts to count, so it will never increment inside an ISR. Since `delay()` uses the same counter as `millis()`, it will not work properly inside an ISR. `micros()` works initially, but will start behaving erratically after 1-2sm. `delayMicroseconds()` does not use any counter, so it will work as normal.

Or would that be too technical/specific?

nickgammon, 2015-08-20:
> but will start behaving erratically after 1-2sm.

Apart from the typo (1-2 ms) that is fine.

> Or would that be too technical/specific? 

I think it's OK. It clarifies things for people who say "it seems to work" when using `micros()` for a short time.

NicoHood, 2015-08-26:
> micros() may jump down during a long ISR because it consists of two parts - the saved counter and the contents of the hardware timer register. If you wait too long the timer register will wrap around twice and then micros() could go down.

@matthijskooijman Maybe this is the reason why I got wrong micros() values inside the ISR some time ago when I was working on the IR library. You remember? I had values BEFORE the last value.

What I wish would be a clear, maybe more advanced explanation when the function will return wrong values. How long the ISR may take at maximum. This is especially needed for my IR library but would help other as well.

agdl, 2015-09-01:
@NicoHood @matthijskooijman and @nickgammon I replaced the statement as suggested. Anything else need to be added? If not close the issue please!

nickgammon, 2015-09-01:
I have to pick a couple of holes in it:

> Since delay() uses the same counter as millis(), it will not work properly inside an ISR. 

Actually delay() uses micros():

```C++
void delay(unsigned long ms)
{
	uint16_t start = (uint16_t)micros();

	while (ms > 0) {
		if (((uint16_t)micros() - start) >= 1000) {
			ms--;
			start += 1000;
		}
	}
}
```

So, not strictly the same *counter* as millis. The same *timer*, yes. I agree it will not work properly, but strictly speaking a short delay would work, in much the same way that micros() works. By "short delay" though, we are speaking of 0 to 1 ms. This might be too much detail for beginners. Maybe:

> Since delay() requires interrupts to work, it will not work if called inside an ISR. 

---

> If your sketch uses multiple ISRs, only one can run at a time, other interrupts will be ignored (turned off) until the current one is finished. 

The other interrupts are not **turned off**, they are deferred. Saying they are turned off is like saying they will never be executed. The interrupts are **not ignored**. They will be executed when the ISR finishes, in interrupt priority order.

---

> To make sure variables used in an ISR are updated correctly, declare them as volatile. 

To make sure variables **shared between an ISR and the main program** are updated correctly, declare them as volatile.

agdl, 2015-09-01:
Done. I changed this statement:

*If your sketch uses multiple ISRs, only one can run at a time, other interrupts will be ignored (turned off) until the current one is finished. *

 as:

*If your sketch uses multiple ISRs, only one can run at a time, other interrupts will be executed in an order that depends on the priority they have.*

nickgammon, 2015-09-01:
> If your sketch uses multiple ISRs, only one can run at a time, other interrupts will be executed **after the current one finishes** in an order that depends on the priority they have.

However if you think that is obvious, then your current wording is fine. :)

agdl, 2015-09-02:
@nickgammon actually as you suggested is clearer for everyone thank you!

TerryKing, 2015-09-03:
Nick, thanks for doing SO much to make Arduino better and clearer!

nickgammon, 2015-09-03:
My pleasure. Cuts down on the question on the forum too. :)

agdl, 2015-09-03:
@TerryKing @nickgammon can we close this? :)

nickgammon, 2015-09-03:
The page has not been updated, however.

https://www.arduino.cc/en/Reference/attachInterrupt

agdl, 2015-09-03:
@nickgammon try now please

nickgammon, 2015-09-03:
Looks good.

closed 2015-09-03 by nickgammon