arduino/Arduino #3701: All code files in variant folders are being selected to compile.

ricklon, 2015-08-19:
All code files in variant folders are being selected to compile. This causes problems when board/variant specific code is being used. It is unnecessary to force these files to compile.

The following code in Compiler.java:
```
  if (variantFolder != null)
      objectFiles.addAll(compileFiles(buildFolder, variantFolder, true,
                                      includeFolders));
```

If that code is removed. Everything compiles properly.

matthijskooijman, 2015-08-19:
This code has a reason, some variants need some code to compile. Can you link to a core or variant that causes problems?

ricklon, 2015-08-19:
Here's the variant folder for the WiFire board:
https://github.com/chipKIT32/chipKIT-core/tree/master/pic32/variants/WiFire

 Board_Data.c includes the files:
```
Board_Data.c
Board_Defs.h
ECADC.cpp
EFADC.cpp
EFADC.h
```
Currently, each of these files is being compiled as if they were stand
alone files. Where they only need to be included into the Board_data.c file.

ffissore, 2015-08-20:
Yep. .cpp files in core folders (including variants folder) are collected recursively, so even putting those files into a subfolder won't work. @cmaglie what do you think?

matthijskooijman, 2015-08-20:
I think these files shouldn't be called .cpp if the intention is not to compile them. And since #include does not care about file extensions, just nameing them ECADC.h, or .cpp.h, or .whatever should adequately fix this I guess?

matthijskooijman, 2015-08-20:
(I wonder why the original post said that removing that bit of java code, which I think means _none_ of the files in the variant folder is compiled, actually works, since then even board_data.c will not be compiled?)

ffissore, 2015-08-20:
I confirm the IDE code looks for `.c`, `.S` and `.cpp` files, so renaming them to `.h` or `.whatever` will work
(I guess that was a quick&dirty solution proposal)

cmaglie, 2015-08-20:
Including a `.c` or `.cpp` file is not a common practice, I guess that the original authors did it this way because at the time the IDE didn't compile all the files in the variant folder, IMHO the solution to rename these files to .h is the correct one (or simply do not include them and they will be compiled anyway).

I've seen this kind of hack also in some libraries, the authors claimed that they "managed to do recursive sub-folder compilation on 1.0.x IDE" (and that's one of the reasons why we implemented the real recursive sub-folder compilation only if the source code is placed in the "src/" folder of the library).

Chris--A, 2015-08-20:
> IMHO the solution to rename these files to .h is the correct one

I agree, please do not remove the ability to include source files. In-fact, although it may be uncommon, including any type of file should be doable. Its not unheard of for code to include things like a `*.csv` to populate an array.

Including a source file is particularly useful when you want to override some mechanism/define implemented in the .cpp. As libraries are compiled before the sketch, any sketch based defines do not affect the library object. By bringing the source to the sketch (include inside namespace), you can have full control over it. Of course this is still a bit of a hack, however it is useful none the less.

ricklon, 2015-08-20:
Cristian,

I'm reconfirming the situation. Including the files inside the Board_Data.c
file might not have been a good practice. Which then would cause this issue.

--Rick

ricklon, 2015-08-20:
Ok. I've confirmed that current behavior is better. Including all the files is proper and the way the file structured was not a good practice. I'm fixing on my end. Thanks for the attention to the problem.

matthijskooijman, 2015-08-21:
@ricklon, thanks for reporting back. I'll close this issue now.

closed 2015-08-21 by matthijskooijman