arduino/Arduino #3702: Add LiquidCrystal_I2C Library

marcoschwartz, 2015-08-20:
Hello,

I would like the LiquidCrystal_I2C library to be added to the Arduino library manager. This library is similar to the LiquidCrystal Arduino library, and adds support for DFRobot I2C LCD screens.

The GitHub repository for the library is:

https://github.com/marcoschwartz/LiquidCrystal_I2C

Thanks!
Marco

ffissore, 2015-08-20:
Added. Thanks!

closed 2015-08-20 by ffissore

Avamander, 2015-11-07:
Would it be possible that this library would be renamed in the Library Manager? This library is not compatible with many I2C LCD modules and sketches using libraries with the same name.

cmaglie, 2015-11-09:
@Avamander 
you (or someone that have push access to the reposiory) can do it:
* remove all the old tags/releases
* rename the library in `library.properties` and create a new release

Once a week we run a "cleanup" task, at that moment the old revisions should be removed from library manager, leaving only the new one.