arduino/Arduino #3704: Add RCReceiver library

willie68, 2015-08-20:
Hi to everybody,

Please, can you add my RCReceiver library to the Library Manager:
https://github.com/willie68/RCReceiver.git

Thanks for your effort!

ffissore, 2015-08-20:
Added. Thanks!

willie68, 2015-08-21:
Hi,

thanks for adding. 
This mornig i just started my IDE and looked into the lib manager. And i don't find the lib? (As the json lib added yersterday too)
When will this be availble? 
I'm working with the 1.6.5r2 Version.

regards 
Willie

ffissore, 2015-08-21:
Just opened lib manager, filtered for "mcs" and there is your lib "MCStools"

willie68, 2015-08-21:
Thanks, now i found it too. 
Just looked for the library name RCReceiver. 
I found that my library.properties is wrong. Should be RCReceiver not MCSTool.
Just my fault.
Thanks for your effort.

closed 2015-08-21 by willie68

ffissore, 2015-08-21:
Remember that, since the wrong name is tagged/released, the only way for you to fix it is by remaking or removing those tags and wait for Sunday, when we rebuild the lib list from scratch rather than adding new released as we do during the week. Or you may just leave them there. Your call

willie68, 2015-08-21:
Thanks, 
it's ok, when the right name appears after sunday. 
So is it sufficient, i only need to make a new release with the right name, and this will be ok after sunday?

ffissore, 2015-08-21:
The **new** release name has just been published. For the wrong name to disappear from **old** releases, you need to delete them or remake them with the right name. Once done, on Sunday, what's currently listed as MCStools 1.3.0 will be listed as RCReceiver 1.3.0

ffissore, 2015-08-21:
It's all git stuff at the end. A bit advanced maybe. But again: you don't **have** to

willie68, 2015-08-21:
Great, working like a charm. 
Again, thanks for your effort.