arduino/Arduino #3709: [REQUEST] Make MAC address parameter optional in Ethernet.begin()

q2dg, 2015-08-21:
If current Ethernet Shields have MAC address fixed from manufacturing, I don't see the need/convenience of specifying it at code level. In fact, I suggest that Ethernet.begin() is more like WiFi.config(): https://www.arduino.cc/en/Reference/WiFiConfig . It would be a very nice change to everybody.

I know older Ethernet Shields (how long older??) hadn't the MAC address fixed...therefore I say it should be maintained as an -optional- parameter.

...also, adding a Ethernet.macAddress() à la WiFi.macAddess() would be a great addition, too

Thanks.

q2dg, 2015-08-27:
More reasons: setting the MAC in software becomes a problem if you want to upload a sketch to multiple Arduinos on the same network. You need to remember to tailor each sketch so that each device has a unique MAC address. If MAC hasn't to be set, this configuration will be independent of the loaded sketch. After initial setup, each Arduino will have its MAC configuration tied to the hardware without the need of changing anything inside the sketch. Subsequently, any number of same sketches can be loaded without problems

agdl, 2015-10-01:
@q2dg the supported chip in the Ethernet library W5100 has no default MAC address, this is the reason why it is initialized in the sketch.. so this request cannot be satisfied I'm sorry..

q2dg, 2015-10-01:
But Ethernet Shields were supposed to be sold with a paper glued to its back where its W5100's MAC was printed. Or that's not the case anymore?

mikaelpatel, 2015-10-01:
@q2dg 
Are you suggesting that the library uses OCR or other method to read the sticker? 

Please explain how you are thinking this would work? There is no extra ROM or EEPROM on the shield so the MCU has a hard time trying to read MAC address on the sticker.

q2dg, 2015-10-01:
Haha, good irony, @mikaelpatel !! Haha!!

Well, I opened this issue because I read this in https://www.arduino.cc/en/Guide/ArduinoEthernetShield#toc2 : 

"Current Ethernet shields come with a sticker indicating the MAC address you **should** use with them. For older shields without a dedicated MAC address, inventing a random one should work, but don't use the same one for multiple boards"

Why then should I use this printed MAC?? I (wrongly) infered that the MAC indicated on sticker was special because it came preconfigured. If it is not, then why there's the sticker? Why don't invent the MAC as  with "older shields" ??

Thanks

mikaelpatel, 2015-10-01:
@q2dg 
The chip (W5100/shield does not allow preconfiguration (no internal EEPROM). It has to be loaded with a MAC address. Any commercial usage would have to use a registered MAC addresses. The sticker is a "recommended" MAC address to use when configuring the device. 

Actually I think this issue is interesting. A solution would be to use a random number instead for configuration. The library should do this internally. This has some advantages but also some disadvantages. For more details see reference 13-14, https://en.wikipedia.org/wiki/MAC_address#cite_note-13.

q2dg, 2015-10-01:
Ok, understood, so it's only a "recommended" address, I'll close the issue then, if you don't mind.
Maybe when/if a "Ethernet Shield 101" comes with a new design, this "problem" could be discussed again.

P.D: References are very interesting, thanks!

closed 2015-10-01 by q2dg