arduino/Arduino #3711: WiFi print line length issue

CountDoozer, 2015-08-22:
If any, individual, line length in a series of client.print statements exceeds ~100 chars the whole page is dropped or sent so badly a browser will ignore it.

Arduino WiFi r3 Version 1.1.0 firmware
Atmega 2650
Arduino 1.6.5 IDE

CountDoozer, 2015-08-24:
The max length is 91 chars including newline.

CountDoozer, 2015-08-24:
Oops, sorry, its 90.

Chris--A, 2015-08-25:
Now that I have my PrintEx library added to the library manager, I can show you how to solve this quite easily.

One of the tools it provides is called `BufferedPrinter`. It allows you to collate lots of small prints to send at once. However a consequence of its design means it can be used to break up super long prints into smaller 'packets'.

```Arduino
#include <PrintEx.h>

//Where ever you use WifiClient....

WiFiClient client = server.available();
BufferedPrinter<90> buffer = client;  //Maximum packet size of 90 bytes.

//Use 'buffer' only for printing. Use 'client' for any reading/client specific operations.
buffer.print( "Hi" );  //Pretend this is a huge string. It may fill the buffer many times.
                      //When the buffer is full it will be written in one go.
                      //Printing 91 bytes will generate two calls to client.write(buf,len).

//Once done, you need to call flush() to ensure any remaining data in the buffer is sent.
buffer.flush(); // << this is different to client.flush() which is part of 'Stream', 
                // maybe I should change its name.

```

I used this with the Ethernet shield as each client.print() call sent a packet. So buffering up a large amount of data, I could send far less packets. Although, it seems you need the opposite. Your Wifi problem has been known for a while, people were discussing it on the forums. See if there is a new firmware for your device. I recall one was released, however I do not know what it changed.

CountDoozer, 2015-08-26:
I must improve my search skills - i had looked for some previous mention of this problem but found none.

PrintEx looks good and is quite a lot like my workaround.  Thanks for pointing it out - I'll look into it.

I'm concerned about all these functions which fail silently.  I spent at least a day on this issue thinking it was something I had done wrong, then coming up with a workaround.

Thanks, again.

Chris--A, 2015-08-26:
No worries.

Here are some topics I found, I haven't read them, but there might be an answer there:
http://forum.arduino.cc/index.php?topic=166151.0
http://forum.arduino.cc/index.php?topic=208913.0
http://forum.arduino.cc/index.php?topic=196030.0
http://forum.arduino.cc/index.php?topic=208913.0

matthijskooijman, 2015-08-26:
FWIW, this sounds like a weird bug somewhere - I wouldn't expect that any *line buffering* happens anywhere by default, wifi data is just a bunch of bytes. I could imagine a limit on a single write call, but it seems weird that multiple print/write calls have a limit on line length.

Good that there is a workaround, but this sounds like a bug somewhere anyway.

@CountDoozer could you perhaps post a minimal example that illustrates the problem (You have described the problem adequatly already, but it would help to verify it also occurs in a sketch that does not do anything else, and it is helpful if the developers have exactly the same sketch for testing)?

CountDoozer, 2015-08-26:
Chris:  thanks for the links (how did I miss this?)
Matthijs: My code is too long, as is, so I'm making a smaller, simpler program and I'll list it here.  It might take until tomorrow.  The original code was working on an ethernet card so I developed it farther than is useful here.  Will scale it back.

CountDoozer, 2015-08-27:
Matthijs: Here is an example:
https://gist.github.com/aba27fa102dfca65c879.git

matthijskooijman, 2015-08-27:
Cool, thanks. I still wonder if the example could be shorter (just printing stuff to each connection, without bothering with all the HTTP stuff, testing with telnet or netcat), though.

I don't actually have the wifi shield to test, anyone else can have a go at reproducing this?

closed 2016-07-12 by agdl