arduino/Arduino #3712: Arduino IDE (Wiring / Sketch) for Linux boards ? ( BeagleBone Black or Raspberry Pi or ATSAMA5D3 Xplained evaluation board ?)  (bump)

VogonJeltz, 2015-08-22:
hey,
after arduino.cc is supposed to have cancelled it's Arduino Tre project (no longer listed on products page -> or will the TRE come soon anyway ? <-), so regrettably the Due's ARM Cortex M3 is currently the upper limit of MCU performance.

Now, will it be possible to implement the Arduino IDE (Wiring / Sketch) for Linux boards, e.g. on BeagleBone Black or on Raspberry Pi ?
Or maybe even on different, more powerful boards feat. more powerful MCUs than just the Due, e.g. at least an ARM Cortex A5 like on ATSAMA5D3 Xplained evaluation board ?
(see: http://www.cnx-software.com/2014/03/07/79-atmel-atsama5d3-xplained-arduino-compatible-open-source-hardware-board-powered-by-sama5d3-arm-cortex-a5-processor/ )

q2dg, 2015-08-22:
I think it is out of the scope of Arduino's mission. They have enough slaughter offering a common approach to so different boards (UNO, Yún, Zero, Due...) which, unfortunately, is not always achieved.
What Arduino should do is facilitate third parties to do what you ask for, and I think it's the idea of Board Manager.

matthijskooijman, 2015-08-22:
Most of those boards run Linux, making programming for them a whole different game. Having said that, I think you could create a custom core yourself (including a platform.txt and boardst.txt) that calls the compiler in the right way to produce either a Linux binary, or a bare-bone ARM binary to run without Linux. This is certainly not going to be implemented by Arduino itself, so I'm closing this issue as wontfix. However if you, or anyone else wants to take this up and it turns out there is something missing in the IDE to facilitate this, just open a new ticket for that and we'll see if that can be fixed.

closed 2015-08-22 by matthijskooijman

PaulStoffregen, 2015-08-23:
Wow, I had no idea the Tre had quietly vanished.

VogonJeltz, 2015-08-23:
Rubbish, I have no idea how to create a custom core and no clue about GCC or ARM binaries. If I knew I woudn't hav esked for that. I'm just a simple stupid Arduino IDE user, and I just use this, nothing else.

peabo7, 2015-08-23:
It's an educational opportunity.

If you try to develop this yourself, armed with all the source code, and run into a roadblock, you can get help from the community.

Don't be a stupid user, get smarter.  You can.

Peter Olson

VogonJeltz, 2015-08-23:
no never. I just use Sketch, nothing else. I would have liked from the heart that the TRE had come, but now I need a more powerful platform as a substitute for the DUE.

If you think you are smart enough instead, do it by yourself!

peabo7, 2015-08-23:
You should hear yourself talk: demand, demand, demand.  I am smart enough to do this, but I am unmotivated by demands.

Other people have tried to explain this to you and you do not listen.  I thought I'd give you another chance.  Too bad, I guess.

The only things I have ever learned about software are from discovering a problem and working until I overcame it.

If you can write a sketch, you have the basic skill of working with C++.  Take it to the next level.  Can you imagine how good you would feel to have a board port you developed?

Peter Olson

VogonJeltz, 2015-08-23:
I have no basic skill for C++, just simple ANSI C.

peabo7, 2015-08-23:
Non answer.

peabo7, 2015-08-23:
Next level.  Get better.

VogonJeltz, 2015-08-23:
no next level, or I just would use the Raspi - but I tried, and failed completely. I'm too stupid for Linux or GCC or such a mess. That's why I just use Sketch.

peabo7, 2015-08-23:
I apologize for not being able to help you.  Good luck in the future.

VogonJeltz, 2015-08-23:
Apalogize accepted. I also wish you the best.

VogonJeltz, 2015-08-23:
But anyway, if the TRE won't come, then a different ARM5 or ARM7 or ARM9 would be needed.

PaulStoffregen, 2015-08-23:
You really seem to have no appreciation for the incredible amount of engineering effort that goes into developing this sort of product, especially good usable software.  Admittedly, Arduino makes these things easy to use and hides all (or at least most) of the difficult details, so there's really no need for you to know such things.

Then again, if you only ever acquire all your food at grocery stores and restaurants, you have no need to know anything of the agriculture industry.  Maybe there too, you'd impatiently demand they produce very new types of products for you to buy and prepared dishes for you to order?

VogonJeltz, 2015-08-23:
peabo7 commented:
"It's an educational opportunity.
If you try to develop this yourself, armed with all the source code, and run into a roadblock, you can get help from the community.
Don't be a stupid user, get smarter. You can."

 matthijskooijman commented :
" I think you could create a custom core yourself (including a platform.txt and boardst.txt) that calls the compiler in the right way to produce either a Linux binary, or a bare-bone ARM binary to run without Linux. "

well, so it sounds the problemshould  be solvable - not just for me as I'm just USING Sketch for ~ 1year, but maybe for any programmer who is a smarter programmer than me and who is experienced in GCC, Linux or ARM binaries or Eclipse or what ever. 
If even I am suppose to resolve this as a beginner by 1 year experience - what I actually can't do nevertheless - then what about the experienced Sketch developers all around here?

PaulStoffregen, 2015-08-23:
> Having a ARM5, a ARM7 or a ARM9 running the Sketch IDE would be great, wouldn't it?

Would it really?

First, it's not even clear to me if you're talking about the Arduino IDE running on this hardware, or the Arduino IDE running on your PC, using such hardware as a super-speed, highly capable board to run the sketches you write.  When you say "ARM9 running the Sketch IDE", taken literally, that seems to mean running the Arduino software on the board, rather than on your PC.  But my guess is you really want a far more powerful board than Uno, Zero and Due?

Simply running sketches faster is of limited value.  Due and Zero already do this, but still Uno is by far the most popular product.  To really be valuable, at least 2 things are needed.

1 - Excellent compatibility.  Faster isn't very useful if the hundreds of widely used Arduino libraries don't work.

2 - Support for special hardware capability.  Even if you can run faster and things work, then what?  If you have hardware capable of multi-channel audio, video, fast networking, etc... how will you use those things?

I believe I can comment on these, having about 3 years of experience developing Teensy 3.x.  After 3 years of a tremendous amount of work, we're finally at the point where nearly all major libs work and several really powerful things like OctoWS2811 and the Teensy Audio Library are mature.  But even on Teensy, tremendous amounts of work are still needed to support USB host and a variety of other hardware capability.

So my point is a half-baked software support effort, only providing the bare minimum functions like pinMode and digitalWrite, would not be particularly great.

But anyone could prove me wrong.  Arduino is a very open platform.  Anyone can create and sell their own boards, using the Arduino IDE and libraries.

If this really would be so great, why hasn't anyone done it yet?

Of course, the answer is because it requires an incredible amount of engineering work.  I believe there are some efforts to run sketches directly on Raspberry Pi.  I'm sure you can find them, if you search.  If you believe just being able to run a sketch on such powerful hardware will be intrinsically great, go find those projects and give them a try?

VogonJeltz, 2015-08-23:
many questions, so one after the other, but pls consider that I am no computer scientist:
I'm talking about using the Arduino IDE to program a MCU it's digital IO ports and communication ports.
If the IDE runs on a PC and generate MCU code by a cross compiler or if the IDE is running on the MCU generating it's own cpu code (interfaced e.g. by a Web IDE to a PC or stand-allone programming by it's own keyboard and an own  HDMI screen . - that does not matter to me.
The TRE was dsesigned to work with a web IDE, the RaspPi or different ARM MCUs like the Lego Mindstorms EV3 ARM9 running Debian Linux (ev3dev) can be programmed and compile both autonomously and by a cross compiler via a PC. If the screen is fine and the cpu power is high, either way would be fine.

So if the web IDE for the TRE was supposed to work fine then, why not do it this way? 
But again, either appoach would be fine for the Arduino IDE, and finally having a substitute for the regrettably abandoned TRE is my primary goal..

I glady appreciate further questions.

VogonJeltz, 2015-08-24:
update:
just discovered that the RaspPi has no ADC ports, so it makes to sense.
Anyway, maybe perhaps BBB or ATSAMA5D3 Xplained evaluation board then.

PaulStoffregen, 2015-08-24:
If it really makes so much sense, wouldn't the people behind those boards do it, so they could sell more?  Or wouldn't someone like yourself (but with the ability to do this) make their own board, perhaps using Kickstarter or Indiegogo to launch it?

systronix, 2015-08-25:
Interesting thread. Another factor is that most everyone also wants a very low price. Many of these projects (Arduino, RasPi, BBB) are subsidized so that the price you see is not an accurate reflection of the engineering effort which created them - not even close. A great example: the Eclipse IDE which is free to use, but was far from free to create. Or Linux. Or alll the FSF products. It seems to me the market of users and creators is still struggling to figure out how to have a sustainable business around such inexpensive, open-source projects. My hat is off to mdeium-size companies like Canonical and small companies like PRJC and Evil Mad Scientist who do not have those subsidies and have to do a huge amount of original work, even leveraging existing infrastructure such as Arduino and Linux.

VogonJeltz, 2015-08-25:
@syreonix:
according to what peabo7 and matthijskooijman wrote it sounds as if it was be quite easy to make, but also different users in our forum assumed the effort to be not overly. According to PaulStoffregen it makes no sense at all. 
I personally am not able to assess the required efforts to implement Sketch on a different ARM 5,7,or 9  board, epecially for this one http://www.cnx-software.com/2014/03/07/79-atmel-atsama5d3-xplained-arduino-compatible-open-source-hardware-board-powered-by-sama5d3-arm-cortex-a5-processor/ 

Anyway, I don't want to get lost in general speculations about free software, mad scientists, ARM binaries, boardst.txt and food at grocery stores and all that, but what people are writing here is quite divergent and I'm not quite sure if all of my previous speakers are really experts about what they are writing about.

VogonJeltz, 2015-08-25:
ps,
what means
"ffissore modified the milestone: Release 1.6.6 an hour ago" ?

matthijskooijman, 2015-08-25:
@VogonJeltz, I think that making _something_ work would be relatively easy, but as @PaulStoffregen said, making it fully and reliably support all of the existing Arduino API and allowing most sketches and libraries to work, will likely need a whole lot more work, if it is even possible at all.

VogonJeltz, 2015-08-25:
ok, then undoubtedly it's clearly no option, regrettably.
Perhaps this was why also the TRE has been abandoned meanwhile.

matthijskooijman, 2015-08-25:
Looking at the original announcement for the TRE, it does not seem like the intention was to make the high-performance part run Arduino sketches directly. Similar to the Yun, it has a 32u4 microcontroller (like the Leonardo), and an ARM CPU running Linux, to be programmed in whatever language or environment is appropriate. I'm just speculating here, though.

VogonJeltz, 2015-08-25:
I'm also just speculating here, but to me it seemed that all the Pins, both the Sitara ones and the AVr ones, could be accessed by the same Sketch IDE, and about multitasking, communication and multimedia (I2C, SPI, UART, HDMI, Audio) the same. 
To have to program both MCUs by different compilers one by one and then had to manage all the communication in between by one self like for the YUN - that indeed had made no sense at all.

mbanzi, 2015-08-31:
Quick clarification: You can already do what you describe with an Intel Galileo and Edison, they both run linux and compile Arduino sketches as linux apps.
The Arduino Tre hasn't been cancelled. It has been paused while we sort out the rest of the product line

VogonJeltz, 2015-08-31:
thank you for clarification!
Galileo and Edison are no option because of too poor number of pins unfortunately, and the Linux of the Galileo is supposed to be really a mess....
I even once tried the Galileo for my Windows XP but did not manage to install anything at all, so I never got it running. I returned it to the provider then.
BTW, I wasn't the only one who desperated about this thing.
http://www.golem.de/news/test-intels-galileo-board-1312-103167-5.html

But about the TRE:
Will it be able to access all GPIOs by 1 IDE which compiles all and everything of both cpus?

q2dg, 2015-08-31:
I really don't wanna be rude and I know it's very offtopic but I must say it: I think there is a really big problem with communication between Arduino Team and community. I know, I know there are urgent and unpleasant legal problems to solve these days but...

*Where is a official note saying TRE is not dead, is only paused until xxx?? Here is the first notice I have about this change of state. The only thing I only knew until now was this board dissappeared one day misteriously as if nobody cared. No news. Any.
*Where is a note saying Leonardo is not dead, is paused until xxx?? (as I remember having read  from you somewhere). 
*Where is a note saying something about what happened to the Yún shield???
*Where is a note saying when will Wifi Shield 101 be available?? Is it so difficult to know?? At least the documentation....

I know you are in a very difficult time, but some little informative notes on web don't hurt and could calm doubts. The general feeling is the project is lurching.

Moreover, another communication problem I detect is the board design decisions aren't justified to community prior to final state and fabrication. For instance: Why Shield 101 lacks microSD slot former Shield had? Why it doesn't has a U.FL (or similar) connector? Etc. And, also, future designs you are on...why don't you open them to discussion? For instance: How do you propose the (supposedly predicted) SAM-D21 breakout ?? Which will be the (supposed) improvement of "new" Leonardo's? Do you think is worthy deprecate GSM Shield in favour of a 3G's one (due to future 2G shutdown) or not??. Etc. Etc.

Thanks, anyway, for all your work and love. All these criticisms are (or want to be) on a positive way.

Sorry if I've bothered someone. I won't do it again.

VogonJeltz, 2015-09-01:
very interesting aspects, but in order  not to get lost in too many technical details I'm actually just curious about the very basics about the TRE:
Will it be able to access all GPIOs by 1 IDE which compiles all and everything for both cpus?

If not, and in case one will have to use Linux console, ssh, putty, native GCC, makefile, or different Linux stuff, the TRE is useless to me anyway, just like the RaspPi or the BBB or the Galileo.

PaulStoffregen, 2015-09-02:
> OTOH, being able to program the whole TRE as simple and easy as the DUE would be amazing.

But only if all sorts of not-well-defined problems are avoided, as encountered with Galileo and Edison.

AND only if you don't read some negative opinions, as with Edison, which prevent you from even trying to use TRE.

@VogonJeltz - You probably are incapable of seeing how unrealistic and nebulous your many requests are.

VogonJeltz, 2015-09-02:
I know, you're number one of demotivators, PaulStoffregen

VogonJeltz, 2015-09-03:
...Massimo Banzi...?

BrentWilkins, 2015-09-08:
@VogonJeltz, you are hilarious...

VogonJeltz, 2015-09-08:
urkomisch
	
hilarious, screamingly funny, killing, riotous, sidesplitting
	
lustig
	
funny, merry, cheerful, amusing, jolly, hilarious
	
ausgelassen
	
hilarious, playful, boisterous, lively, frisky, rollicking
	
übermütig
	
cocky, wanton, high-spirited
	
sehr komisch
	
hilarious

.........?

mbanzi, 2015-09-08:
@q2dg
I dunno if you followed the recent Arduino issues but we have been a bit busy with a few legal issues that are annoying but tend to waste our time.
In addition to this we have been also busy re-booting our manufacturing efforts. We're now making Arduino boards in the USA for the US market and shipping more and more every day. 

Looking at our product line we decided to pause some efforts and to remove some products because we prefer to concentrate on some products moving forward. 
I believe I've discussed this and the logic behind it in a message on the developer mailing list.

This is not the forum to discuss Arduino products and we don't want to spam the rest of the devs with this.

At maker faire NY i'll provide an update on many of the items you bring up

mbanzi, 2015-09-08:
@VogonJeltz please do not spam this thread. I took the liberty to remove your comment as it didn't add anything to the conversation.

Paul is incredibly competent and solid. Everyting he writes is very true. Calling him a " defeatist and demotivator" doesn't do any justice to the amount of contributions he made to the Arduino project.

In the TRE project we have two implementation of the Arduino API, one running on the Atmega32u4 which can be used for time-sensitive tasks where a native implementation is more relevant and an implementation running on top of the Sitara processor in the form of a linux process. This gives you access to the GPIO and busses on the board. so you have 2 ways of running Arduino code depending on what kind of task you're trying to accomplish.

VogonJeltz, 2015-09-09:
thanks, but my question was about projects which aquire both all the Sitara GPIOs and ports and the AVR GPIOs and ports plus the computation and memory power of the Sitara. So I expected to access all pins and all funcionality by 1 IDE by 1 program code. Will that be possible or not ?

meistro57, 2015-11-27:
Have you heard of the PcDuino? It's not as fast as BBB but has an Arduino kinda built into it. Try the Chipkit boards if your looking for speed.

VogonJeltz, 2015-11-27:
yes, sure I heard about it, but can it be programmed by the Arduino IDE?
And does it suppport GPIO access and Timer interrupts on hardware level, not just by Linux user-space (that's also the crucial point for the Raspberry PI...!)?

meistro57, 2015-11-28:
It has the Arduino IDE on the desktop. Write a sketch with it to control
the Arduino compatible pins on the board.

VogonJeltz, 2015-11-29:
seems to require lot of Linux stuff and skills though...
I need just a simple cross compiler for the PC, like the Arduino IDE, no Linux:

write programs on the PC
compile on the PC
upload to the Arduino board
run the program on the Arduino and debub via the serial console

meistro57, 2015-11-29:
?? Really? Then what's wrong with Arduino? If it's speed your after then
get yourself a Chipkit board. They have pic32 and if you can't get enough
speed from 80mHz then WTF are you trying to do with it?

I'm an ironworker. I got my 1st Arduino right when they first came out and
had 9 pin serial plug on it. I had to develop skills to stay entertained
with this stuff. Yes it's challenging. I could have spent some serious $$
and had someone else do it all for me. Anyone can do that. If your
motivated enough then you'll figure out somehow how to do whatever it is
you want to do. I have skills now because I did my own work. Yes, I asked
for help. But I made it clear that I wasn't asking anyone to solve my
problems for me. I asked for help to understand how it worked and learn
why. I have skills to build CNC machines in my garage because of Linux
stuff and Arduino. If I can do it then anyone can.

If your looking for the easy way then your looking in the wrong place.

 Go buy a finished product that you understand nothing about because you
never took the time to learn some skills.

VogonJeltz, 2015-11-30:
the topic was about the Arduino TRE, featuring the Arduino IDE for all the plugs and GPIO pins.

The Due is not usable unfortunately because it has too poor memory (the TRE was supposed to have 512 MB, the Raspi has up to 1 GB, but at least 64 MB  are required).

The Raspberry Pi, as I just encountered, does not provide hardware-controlled GPIO pins (just by the unpredictable and unreliable user-space) and far too few pwm pins.

That's why I asked for the TRE or a coequal substitute.