arduino/Arduino #3714: Global programmers.txt file

jsleroy, 2015-08-22:
Hi,

Not sure if this is a bug or an enhancement request. As well, please excuse me if this topic is already covered somewhere or solved using a technique I don't actually know.

This request is for Arduino IDE > 1.6.4.

I've been trying to use an ISP programmer from Tux Graphics, which is basically a stk500v2.
In order to make it work, I had to hack the installed programmers.txt file. The changes are:

avrispmkii.communication=serial
avrispmkii.speed=115200
avrispmkii.program.protocol=stk500v2
avrispmkii.program.speed=115200
avrispmkii.program.extra_params=-P{serial.port} -b{program.speed} -c{program.protocol}

I've read the wiki to see how to add 3rd party support. So I've been trying to add this programmer, so that I don't have to hack released files, and get something stable with time/releases.

So I've add a new programmers.txt file in ~/Arduino/hardware/arduino/avr/.

The content is:

tuxgraphics.name=Tux Graphics
tuxgraphics.communication=serial
tuxgraphics.protocol=stk500v2
tuxgraphics.speed=115200
tuxgraphics.program.protocol=stk500v2
tuxgraphics.program.tool=avrdude
tuxgraphics.program.speed=115200
tuxgraphics.program.extra_params=-P{serial.port} -b{program.speed} -c{program.protocol}

However the IDE is issuing an error message:

WARNING: Error loading hardware folder /home/xxx/Arduino/hardware/arduino
  No valid hardware definitions found in folder arduino.

I guess it's caused by the fact that I've just defined programmers, not boards in boards.txt nor platforms. I've also tried moving the file in ~/Arduino, but that doesn't seem to work.

So here come my request. Would it be possible to allow to define new programmers, without having to define a whole new 3rd party dirextory/files (because usually you might want to add other programmers while still keeping default released boards and platoforms).

Best regards,
Jean-Sébastien

matthijskooijman, 2015-08-22:
Seems like this is a valid usecase. I believe that this is also possible for boards.txt, though I am not entirely sure.

NicoHood, 2015-08-23:
You need to put the file into `sketchbook/hardware/mylib/avr`
Also a platform.txt is required, but there does not need to be much inside.

Example:
https://github.com/NicoHood/HoodLoader2/tree/fast-usbserial

jsleroy, 2015-08-23:
I've tried your suggestion, i.e. adding a `programmers.txt` in `/home/.../Arduino/hardware/tuxgraphics/avr` with a `platform.txt`. However the IDE (1.6.5) is still complaining about missing `board.txt`.

The content of `platform.txt` is copied from  https://github.com/NicoHood/HoodLoader2/tree/fast-usbserial:

`name=Tux Graphics`
`version=1.0`

I wonder if I'm missing something.

NicoHood, 2015-08-23:
is this your sketchbook folder or your .arduino15 folder?
Try to add an empty boards.txt file. Is your work somewhere on github?

jsleroy, 2015-08-23:
@NicoHood The files are in my sketchbook folder (which is actually `~/Arduino`).

Adding an empty `boards.txt` and adding a bit of information in `platform.txt` solved this issue.
Many thanks !

I've created a repository if you want to have a look at the files.
https://github.com/jsleroy/arduino-tux-graphics

closed 2015-08-23 by jsleroy

NicoHood, 2015-08-23:
I think the platforms.txt only needs the name and version. Is this really needed?
https://github.com/jsleroy/arduino-tux-graphics/blob/master/hardware/tuxgraphics/avr/platform.txt#L4-L10

jsleroy, 2015-08-23:
Not sure it's needed.
Actually without these lines I get the following issue:
`Error while uploading: missing 'program.params.verbose' configuration parameter`

Adding only `program.params.verbose` I get a Java Exception from the IDE.