arduino/Arduino #3715: home_and_end_travel_far no longer works as of 1.6.5

beargun, 2015-08-22:
On Mac OSX, by default, the home and end keys travel to the begin and end of the document.  editor.keys.home_and_end_travel_far=false, inside preferences.txt, changed this to a more useful begin and end of line.

This worked till version 1.6.4 , but no longer does anything as of 1.6.5

ffissore, 2015-08-24:
Fixed. Fix will be available in next hourly build http://www.arduino.cc/en/Main/Software#hourly
In order to avoid unsolicited changes in behaviour, the pref in now called `editor.keys.home_and_end_beginning_end_of_doc`

beargun, 2015-08-27:
You used `home_and_end_to_start_end_of_doc` in the code, but `home_and_end_beginning_end_of_doc` in the preferences.

Also in setting `home_and_end_to_start_end_of_doc` to true you would think **home and end** would make you go **to** the **start** and **end of** the **doc**ument, but it's actually the other way around. Totally backwards.

The code itself works like a charm. I'm really happy.

ffissore, 2015-08-28:
Sorry about that. Fixed. Check out next hourly

commit fa78f718 (2015-08-24): Editor on MacOSX: editor.keys.home_and_end_beginning_end_of_doc, when set to true, makes HOME and END keys go to the beginning and end of the line. Fixes #3715
closed 2015-08-28 by ffissore