arduino/Arduino #3718: [DOC] Clarification of Stream methods' availability  in objects from derived libraries

q2dg, 2015-08-22:
Hello. I read in https://www.arduino.cc/en/Reference/Stream that Stream object is the base of several other objects (namely, five: Serial, SD, Wire, EthernetServer and EthernetClient) and it has several methods (find(), findUntil(), readBytes(), readBytesUntil(), readString(), readStringUntil(), peek(), parseInt(), parseFloat(), setTimeout()...) which are only mentioned there and in Serial's page (https://www.arduino.cc/en/Reference/Serial) but not in the other four ones (SD, Wire, EthernetServer, EthernetClient). So, it isn't clear if it is possible to use all Stream methods as SD/EthernetServer/etc methods. I mean, for instance,could I write this?

#include <SPI.h>
#include <Ethernet.h>
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED }; 
byte ip[] = { 192, 168, 1, 177 };                                       
EthernetServer myserver(23); 
EthernetClient myclient;       
boolean found;
void setup() {
	Ethernet.begin(mac, ip);                                    
	myserver.begin();                                             
	Serial.begin(9600);
}
void loop() {
	myclient = myserver.available();   
	if (myclient > 0) {                              
		found=myclient.find("astring");
		Serial.println(found);
	}
}

It compiles but I have'nt right now a Ethernet Shield to check if this sketch works as it should: it isn't clear from documentation. 

In conclusion: I think Stream and derived/related pages need a big rewrite to clarify/arrange this.

Thanks.

q2dg, 2015-08-22:
Well, it seems WiFi library also derives from Stream because it has several familiar methods: available(), read(), flush()...but this suspiction is not asserted anywhere.

per1234, 2017-07-03:
@q2dg had a good comment on https://github.com/arduino/Arduino/issues/2919 that might otherwise be missed since that issue was closed as a duplicate of this one:
>I think documentation should be refactored to list in one point common methods (Stream page) and then, in library reference pages a link to these (cleaning some redundancies).

I think this would make the reference pages easier to maintain since it would avoid a lot of duplicate content.