arduino/Arduino #3719: Sketch upload hangs on certain Serial.println text on a Mega2560

pepijn1978, 2015-08-23:
If I try to upload this sketch below, the Arduino hangs:

void setup() {

  Serial.begin(115200);
  Serial.println("EEPROM geschreven met P1_date !!!!");

}

void loop() {

}

Any other text for Serial.println will do fine.

The sketch only hangs on my Mega2560. Not on my Uno.

Chris--A, 2015-08-23:
Sending three `!` during upload triggers the bootloader's device info tool. You can either print it in two commands:

```C++
Serial.print("EEPROM geschreven met P1_date !!");
Serial.println("!!")
```

Or you can upgrade your bootloader, there is a newer version which does not include the feature triggered by `!!!`.

matthijskooijman, 2015-08-23:
Have a look at http://gammon.com.au/bootloader for an easy way to upgrade the bootloader. Since this is a known and fixed issue, I'm closing this.

closed 2015-08-23 by matthijskooijman