arduino/Arduino #3725: Add Example Group title

00alis, 2015-08-25:
We are redesigning the Tutorial pages on arduino.cc and it would be super helpful if we could add a title to each Example group in the menu (as we did for Boards, attachment)
[image]

Specifically the titles would be:
- Built-in Examples
- Examples from Libraries
- Examples from Custom Libraries

If this copy is too long we can consider removing the word Examples everywhere.
Thanks @ffissore !

matthijskooijman, 2015-08-25:
I implemented pretty much this in #1986, see https://github.com/arduino/Arduino/pull/1986#issuecomment-41328050 for a screenshot. That PR is probably severely outdated (and this is not the main goal of the PR), but it can still probably useful. If the rest of the PR is also considered useful, I could have a go at updating it.

NicoHood, 2015-08-26:
Since we now have the categories I agree with @matthijskooijman to add category folders as well as the 3 main categories as described above. My library list is very long even though I have not installed that ultra much libraries. Especially the builtin number of libs is very long.

commit de8fcc9e (2015-09-21): Examples menu: added group title. Fixes #3725
closed 2015-09-21 by ffissore