arduino/Arduino #3726: Arduino Editor: jerking lines for mose mark/ drag and

VogonJeltz, 2015-08-25:
when putting the cursor in one of the upper or lower lines in the editor window e.g. for marking a code block (keeping left mouse button pressed, then drag the mouse)  the lines jerk and jump up and down, this makes handling of draging very complicated.

Additionally, if code block (e.g., a few lines) is already marked, draging this block to another position is not possible.

Fixing these issues would be appreciated!

ffissore, 2015-08-26:
Can you screencast a video? I can't reproduce both the issues: selection works fine, moving a selected code block using the mouse works fine

VogonJeltz, 2015-08-26:
no, sorry, I cant make a video unfortunately.
I'm using IDE 1.6.3 for Windows XP32.

about moving a marked block:
when I drag the mouse with pushed-down left btn, I can mark.
But when I then put the mouse pointer on the marked area, the markes area dissapears.so I can't move it.
With MS Word it works.

About the jerking:
Put the mose pointer on the 2nd line from above => the lines jerk down.
On the bottom it's vice versa, it jerks up.

ffissore, 2015-08-27:
Not even a screenshot? As I said, I cannot reproduce those issues

VogonJeltz, 2015-08-27:
a screenshot of a disappearing selection:
1) clicking into the line 10:
[image]
2) holding the mouse pressed the lines jerk down and everything is going to be unintentional selected until up to line 1:
[image]

about mark/select and the try to move the highlighted block:
1) select a block:
[image]
2) now trying to click into the selected heighlighted area in order to move it  
=> selection disappears at once, no moving/draging possibe!
[image]

Chris--A, 2015-08-27:
You'll have to update your IDE to see if this still exists.

The editor was replaced with RSyntaxTextArea in 1.6.5, your screenshots are using 1.6.3

ffissore, 2015-08-27:
@Chris--A good catch!! Sorry @VogonJeltz, you also wrote that in your first reply, but somehow I missed it. Closing as worksforme then. @VogonJeltz upgrade to 1.6.5 and reopen if your issues are still valid with the latest version

closed 2015-08-27 by ffissore

VogonJeltz, 2015-08-27:
yes, I already wrote that I'm using 1.6.3.
If the issue has been fixed meanwhile then it's fine of course.

I wouldn't update to 1.6.5 though because the boards and the library manager is too weird, I can't update all my installed boards, all my existing libs and all my lib.c float patches without huge issues then.