arduino/Arduino #3727: Arduino Editor: Search and/or  Replace

VogonJeltz, 2015-08-25:
IMO there should be an improved function for different 
search
and
search+replace
operations
providing different settings for noting
- upper/lower case
- complete words only 
- searchin direction
- and some more settings (screenshots taken from Bricx Control Center)

[image]

[image]

hashhar, 2015-09-01:
Could you mentor this for me? I want to contribute to the project but am pretty new to open source maintainance. Let me know about the files which handle this and I'll look into it.

ffissore, 2015-09-08:
@hashhar file is cc.arduino.view.findreplace.FindReplace

hashhar, 2015-09-08:
Okay. I will start working on it. Thank you.

hashhar, 2015-10-04:
I don't think I can do this in a reasonable amount of time. I am utterly confused. I guess I'll read up and try later. Also, new to Java.

ffissore, 2015-10-05:
Thank you all the same

VogonJeltz, 2015-10-05:
2 features of the above actually would be needed:
1) seperate menus for search and replace (which should not be too hard to do if one is experienced in Java
2) a additional option "whole words only" as the current S+R functon muddles everything up by relacing all and everything even inside of words and expressions.

hashhar, 2015-10-06:
I think I can improve the search and replace algorithm to match whole words only. I'll try and do it.
As for the F3 follow up, its out of my current Java knowledge set.
Separate menus only need a GUI re-write which I got really confused by (I'm more of a .NET and Visual Studio guy)

lmihalkovic, 2016-01-27:
actually all that's missing is REGEX ( not sure i want to do search within selection only for now)

[image]

see #4488 for more

digimeas, 2016-06-20:
Hi!
I have no problem with the current FindReplace options, but maybe;
 * put the Case/Wrap/Tabs checks all on the same line to save space?  Plleeaassee?
 * don't pop the dialog back into the middle of the window covering half the text after I've very carefully positioned it where I want it out of the \<ahem\> way because it's so  \<ahem\> HUGE?   Plleeaassee?

Thanks.  (Like you guys don't have better things to do :-)
Doug

tofrnr, 2018-01-03:
is there meanwhile an update about the extra option "search whole words only" ?
without that option even the "replace all" button causes total mess!

shiftleftplusone, 2018-11-16:
when will there finally exist an improved search / find / replace function in the IDE, feat. also searching just for whole words optionally?

dsyleixa, 2019-01-30:
hello,
are there any news and updates about this issue?

> 
> @ffissore ffissore added the Type: Improvement label on 26 Aug 2015
> @ffissore ffissore self-assigned this on 26 Aug 2015

dsyleixa, 2019-05-06:
another 3 months elapsed: 
still no ETA for this feature?
what about @ffissore self-assignement?

matthijskooijman, 2019-05-06:
I suspect there are a lot of other things with more priority. If anyone wants to implement this, things could move along more swiftly, of course.

As for @ffissore, he no longer works for Arduino.

dsyleixa, 2019-12-16:
this missing "whole words only" thing is really totally bunkers - even minmalistic search tools have that option!
And same about "Find" (ctrl+F) versus an extra "Search and Replace" (ctrl+H)!