arduino/Arduino #3736: open *.h files with the Arduino IDE - why not possible?

VogonJeltz, 2015-08-27:
Trying to open *.h files with the Arduino IDE - why is this not possible?
I need to revise my own .h header file for my current project, but this is not possible incomprehensibly.

PaulStoffregen, 2015-08-27:
Well, it's a good thing you describe these issues so clearly, with such well defined steps to reproduce the problem.  Nothing said "incomprehensibly" here!

VogonJeltz, 2015-08-27:
my English is quite poor - so I admittedly don't understand quite clearly what you want to express with your post...?

per1234, 2015-08-28:
Put a blank .ino file in the folder with the same name as the folder(in this case ardustdio.ino) and then open that file, all .h and .cpp files in the folder will also open as tabs. I use the .ino file as a place for notes. If you want to edit files in the libraries folder then see this: https://github.com/arduino/Arduino/issues/3512

VogonJeltz, 2015-08-28:
I know that there are workarounds and I already had to use them, my point is to make it possible for the future also for  all
*.h
*.cpp
(*.c)
files to be opend and saved and "saved as" by either extension just as if they were *.ino

ffissore, 2015-08-28:
The reason is that the Arduino IDE is made to edit/compile/upload sketches. It's not notepad

closed 2015-08-28 by ffissore

VogonJeltz, 2015-08-28:
I know that it's currently made this way, but to have to use any workarounds for that simple task is incomprehensible and unreasonable IMO nevertheless.

Please change it anyway!!!

lmihalkovic, 2015-10-31:
Ran into the same issue. Made some IDE changes to be able to include libraries locally inside a sketch folder. These local libraries are editable like the rest of the code in the main sketch folder.
https://github.com/arduino/Arduino/issues/4050

LindsayFowler, 2019-03-27:
Saying Arduino is not Notepad is not an excuse. I too need to see and edit the header files in my project and I can't because of this silly feature. It's annoying to have to open them in Visual Studio.

deddicarles, 2019-04-22:
if you want to edit or open the ".h" and ".cpp" file from arduino library, just open it with notepad and copy paste to the arduino IDE and you can save it as sketch file (".ino") or save again as ".h" and ".cpp" as library.