arduino/Arduino #3738: LoadVIDPIDSpecificPreferences fails when connected board returns no string text

logan893, 2015-08-27:
I'm receiving the following error with the newly downloaded Arduino IDE 1.6.5-r4, when compiling any sketch.

java.lang.NullPointerException
	at cc.arduino.LoadVIDPIDSpecificPreferences.load(LoadVIDPIDSpecificPreferences.java:48)
	at processing.app.debug.Compiler.createBuildPreferences(Compiler.java:592)
	at processing.app.debug.Compiler.<init>(Compiler.java:193)
	at processing.app.debug.Compiler.build(Compiler.java:96)
	at processing.app.Sketch.build(Sketch.java:1109)
	at processing.app.Sketch.build(Sketch.java:1084)
	at processing.app.Editor$BuildHandler.run(Editor.java:1955)
	at java.lang.Thread.run(Thread.java:745)

I need to have my Arduino Pro Mini board (most likely a Chinese knock-off) connected via USB and have that COM port selected for this error to show up. The port shows up as e.g. COM4, with no additional parenthesis like my Arduino Uno which shows up as COM7 (Arduino Uno).

Disconnecting this device (i.e. selecting none), or connecting and selecting my Arduino Uno, allows me to compile.

Source file at fault:
https://github.com/arduino/Arduino/blob/1.6.5-r4/arduino-core/src/cc/arduino/LoadVIDPIDSpecificPreferences.java

This file is new as of 1.6.5-r4

Line 48:
    int VIDPIDIndex = findVIDPIDIndex(prefs, boardPort.getPrefs().get("vid").toUpperCase(), boardPort.getPrefs().get("pid").toUpperCase());

One, or likely both, of the boardPort.getPrefs().get(String) functions returns null, which throws this exception when calling .toUpperCase() on a null value (default return value from java.util.LinkedHashMap.get() when no match is found).

Possible suggestion to fix this (you likely know better if this has any other side effects); use a default option for the get() function of PreferencesMap data structure.
    int VIDPIDIndex = findVIDPIDIndex(prefs, boardPort.getPrefs().get("vid","").toUpperCase(), boardPort.getPrefs().get("pid","").toUpperCase());

cmaglie, 2015-08-27:
Thanks for the quick and detailed report!
We'll check this out ASAP.

haaland1, 2015-08-27:
I am having the same issue. I downloaded version 1.6.5 of Arduino IDE this morning and receive the same error when compiling code with my Arduino Pro Mini connected via USB.

java.lang.NullPointerException
	at cc.arduino.LoadVIDPIDSpecificPreferences.load(LoadVIDPIDSpecificPreferences.java:48)
	at processing.app.debug.Compiler.createBuildPreferences(Compiler.java:592)
	at processing.app.debug.Compiler.<init>(Compiler.java:193)
	at processing.app.debug.Compiler.build(Compiler.java:96)
	at processing.app.Sketch.build(Sketch.java:1109)
	at processing.app.Sketch.exportApplet(Sketch.java:1127)
	at processing.app.Sketch.exportApplet(Sketch.java:1113)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2380)
	at java.lang.Thread.run(Thread.java:745)

The sketch compiles fine when the board is disconnected.

mojtabacazi, 2015-08-27:
same here
```
java.lang.NullPointerException
	at cc.arduino.LoadVIDPIDSpecificPreferences.load(LoadVIDPIDSpecificPreferences.java:48)
	at processing.app.debug.Compiler.createBuildPreferences(Compiler.java:657)
	at processing.app.debug.Compiler.<init>(Compiler.java:214)
	at processing.app.debug.Compiler.build(Compiler.java:99)
	at processing.app.Sketch.build(Sketch.java:1117)
	at processing.app.Sketch.exportApplet(Sketch.java:1135)
	at processing.app.Sketch.exportApplet(Sketch.java:1121)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2421)
	at java.lang.Thread.run(Thread.java:745)
```

Juanah, 2015-08-27:
downgrade to 1.6.4 ;)

objectiveSee, 2015-08-27:
Oh geez I guess I'll downgrade too!

netstv, 2015-08-27:
I get it on the nano as well.. downgrading...

PaulStoffregen, 2015-08-27:
I'm also seeing the bug here.  :(

> downgrade to 1.6.4 ;)

If you're going to downgrade, you'll probably want to skip 1.6.4 and go back to 1.6.3, so you don't get hit with issue #3115.

netstv, 2015-08-27:
I don't seem to have that problem.  (Yet)

dmjlambert, 2015-08-28:
Perhaps 1.6.5-r2 could be posted on the website download section again, since there have been several issues brought up in the forum about this.  New users are having trouble.

closed 2015-08-28 by ffissore

cmaglie, 2015-08-28:
The fix is available for download as 1.6.5-r5:

https://www.arduino.cc/en/Main/Software

Joan-Chen, 2015-09-02:
I'm a newer here, just download 2015/8/28,  Arduino 1.6.5,  and got the same exception error within the example file : Blink ,  that means it will be better to re-down load the older version?

cmaglie, 2015-09-02:
Just download the latest version 1.6.5 again, it should be named arduino-1.6.5-r5-windows.exe. This release with **-r5** has the bug fixed.

Joan-Chen, 2015-09-02:
Thanks, will try again.

manjunath-m, 2015-09-08:
Release 1.6.4 just worked fine. I had similar issue with my MinimOSD board with ATmega328P using Arduino 1.6.5

ffissore, 2015-09-09:
@manjunath-m 1.6.5 works fine. Your issue has nothing to do with this one. Please contact MinimOSD developers

Joan-Chen, 2015-09-09:
Great, had down load , will try 😃

manjunath-m, 2015-09-09:
@ffissore 1.6.4 worked for me. So cheers!

Joan-Chen, 2015-09-11:
now, using the 1.6.5,the compiler seems ok, but,  there is a message like 

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x20
.......
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30 ....

I use the sample prog. Blink
what is that means?

Joan-Chen, 2015-09-15:
base on no answer back, I try another board,  NANO, using another computer in WIN7,  first it can not be installed on USB 3.0,  so, exchange to USB 2.0,  then  it's even worse to connect to,  but seems to be the same problem, as the menu show up the com port as COM1,  but,  when upload the code,  it return the message,  no COM1 connect ~   OMG!   should I take an elder version?  for the WIN 7 SERVICE PACK 1?  it means my board out of order?

commit f2427b47 (2015-08-28): LoadVIDPIDSpecificPreferences was giving for granted that the attached board provided VID & PID. This is not always true, especially with counterfeits. Fixes #3738