arduino/Arduino #3740: Executing the arduino in console mode does not work when building non-Arduino packages

fallberg, 2015-08-27:
When the arduino IDE is invoked in command line mode to compile a sketch, it does not reference the proper libraries.
When executed in GUI mode it does compile properly.
Either there is a "secret" argument that needs to be passed for the library manager to search in the correct location for a given package, or this is a bug:
```
/opt/arduino-1.6.5/arduino --verify -v --board esp8266:esp8266:generic --pref sketchbook.path=<path to sketchbook> <path to sketch>
```
**Result:**
```
Using library SPI in folder: /opt/arduino-1.6.5/hardware/arduino/avr/libraries/SPI
WARNING: library SPI claims to run on [avr] architecture(s) and may be incompatible with your current board which runs on [esp8266] architecture(s).
```

ffissore, 2015-08-28:
Are you using SPI library? Do you have an SPI library for your esp8266 sketch?

matthijskooijman, 2015-08-28:
@ffissore, wasn't the problem that if --board specifies a board in a different architecture/core, the core-specific list of libraries does not get re-read? I recall an earlier bug report about this as well, I think that was #3503 (making this a duplicate).

fallberg, 2015-08-28:
@matthijskooijman @ffissore Yes, exactly. I have specified a --board structure. And it is not AVR. So why is AVR libraries included? And yes, I do have all necessary libs, since it works just fine when I use the IDE in GUI mode and select the generic board for that architecture in the menu and compile. Then the correct libraries are referenced.

From the GUI console:
```
Using library SPI in folder: /home/fallberg/.arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/SPI
```
The esp8266 package is installed from here: https://github.com/esp8266/Arduino

However, I have noticed something interesting: When I have opened and compiled using the GUI, it suddenly works to compile from console (with the same arguments that did not work before). So the GUI set some magical properties when executed that the console version inherits somehow.

fallberg, 2015-08-28:
Unfortunately, I now see that subsequent calls to the console "version" sticks to the esp8266 libs when I actually wants to target an Arduino board. So there is something "sticky" from the GUI when picking a board.

fallberg, 2015-08-28:
By manually editing .arduino15/preferences.txt and changing the target_package and target_platform properties, I can build. But this cannot be how it is supposed to work.

fallberg, 2015-09-07:
Any update on this @ffissore? I am not sure what more feedback is expected from me.

matthijskooijman, 2015-09-07:
@fallberg nothing, I think. The next step is that someone sits down, looks at the code and fixes this. I think the problem is clear enough. If you want to dig into the java code for this, you're welcome of course :-)

sandeepmistry, 2015-09-07:
@ffissore this is what I was running into as well when switching from building for a AVR board to a SAMD board.

https://github.com/arduino/Arduino/pull/3759 should resolve this.

fallberg, 2015-09-07:
@matthijskooijman if I knew java I might, but unfortunately I am an old mouldy C developer :)
If anything I would then like to code away all horrible dependencies to an X server in the Linux port. That bug bothers me the most since I am running builds on a headless CI server. 
@sandeepmistry that sounds great! I will keep an eye open for that PR then and hope it makes the 1.6.6 release.

ffissore, 2015-09-08:
@fallberg #3759 has been merged and it's now available with the hourly builds http://www.arduino.cc/en/Main/Software#hourly
Can you download and try it?

fallberg, 2015-09-08:
@ffissore Yes. It seem to work better now. I do get this but it compiles and links.
Invalid library found in /home/fallberg/.arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/esp8266: /home/fallberg/.arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/esp8266
Invalid library found in /home/fallberg/.arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/esp8266: /home/fallberg/.arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/esp8266
Verifying...
Using library SPI at version 1.0.0 in folder: /home/fallberg/.arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/SPI
Using library MySensors in folder: /var/lib/jenkins/jobs/MySensorsArduinoPR/workspace/libraries/MySensors (legacy)
Using library EEPROM at version 1.0.0 in folder: /home/fallberg/.arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/EEPROM
Using library ESP8266WiFi at version 1.0.0 in folder: /home/fallberg/.arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/ESP8266WiFi

fallberg, 2015-09-08:
The folder /home/fallberg/.arduino15/packages/esp8266/hardware/esp8266/1.6.5-947-g39819f0/libraries/esp8266 contains examples. Perhaps incorrect use within a library package but I did not see those "invalid" messages before.

ffissore, 2015-09-09:
Good to know. Stuff under "libraries" is supposed to be a library. That [`examples`](https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266/libraries/esp8266/examples) folder is just a folder, hence the warning

closed 2015-09-09 by ffissore