arduino/Arduino #3741: no issue but maybe may include this lib if he likes it

VogonJeltz, 2015-08-28:
~~~
// Library: ardustdio.h
// extended C-string and kind of stdio.h functionality for files
//
// (C) Helmut Wunder (HaWe) 2015
// freie Verwendung für private Zwecke
// für kommerzielle Zwecke nur nach Genehmigung durch den Autor.
// Programming language: Arduino Sketch C/C++ (IDE 1.6.1 - 1.6.3)
// protected under the friendly Creative Commons Attribution-NonCommercial-ShareAlike 3.0 Unported License
// http://creativecommons.org/licenses/by-nc-sa/3.0/   //   

#ifndef  ARDUSTDIO_H
#define ARDUSTDIO_H

/*

long     fprintf_( File * stream, const char fmtstr[], ... );       // see ANSI C: fprintf()
long     fscanf_ ( File * stream, const char fmtstr[], ... ) ;      // see ANSI C: fscanf()
char   * fgets_  ( char * str, int32_t num, File * stream );        // see ANSI C: fgets()
File     fopen_  ( char * filename, const char * mode );            // see ANSI C: fopen()
int16_t  fclose_ ( File   file_);                                   // see ANSI C: fclose()   
int16_t  remove_ (char * filename);                                 // see ANSI C: remove()
 

char  * ftoa(char * str, double f, int16_t precision );             // converts float to string by precision (digits)

char  * strinsert( char * source, char * sub, int pos );            // insert a substr into a string at pos
char  * strdelposn(char * source, int16_t  pos, int16_t  sublen);   // delete a substr in string at pos
char  * strpatch ( char * source, char * sub, int pos );            // patch string by substr at pos
char  * substr  (  char * source, char * sub, int pos, int len );   // get substring of source string at pos by length
int16_t strchpos(  char * str, char ch );                           // find 1st occurence of char ch in string str
int16_t strstrpos( char * haystack,  char * needle)                 // find 1st occurance of substr in str
char  * stradd(char * s, int n, ...)  // "adds strings" s=sumstring , n=number_in_list, ... = var string list

*/

#define fileIO_OK            +1
#define fileIO_NO_ERR         0
#define fileIO_ERR_CREATE    -1
#define fileIO_ERR_OPEN      -2
#define fileIO_ERR_REMOVE    -3
#define fileIO_ERR_WRITE     -4
#define fileIO_ERR_READ      -5
#define fileIO_ERR_IMPLAUS   -6
#define fileIO_ERR_NAME      -8
#define fileIO_ERR_SDCARD   -16

//********************************************************************************************

int32_t  fscanf_ ( File * stream, const char fmtstr[], ... ) {         // see ANSI C: fscanf()
   const  int32_t   MAXSTRSIZE = 1024;
   char   str[MAXSTRSIZE];
   va_list   args;
   int32_t   i=0, cnt=0;
   int16_t   chr, argcnt=0;
   char      argstart, ch;
 
   va_start(args, fmtstr);
   strcpy(str, "");
   while ( (stream->available()) && (i < MAXSTRSIZE-1) ) {   
      chr = stream->read() ;
     
      if (chr>=0 && chr!='\n') {  // additionally limit to number of arguments! <<<<<<<<
           ch = (char)chr; 
           str[i] = ch;     
           ++i;
      }
      else break;     
   } 
   str[++i] = '\0';                                                         
[... 193 lines cut ...]
}

//********************************************************************************************

char * stradd(char * s, int n, ...)  // "adds strings" s=sumstring , n=number_in_list, ... = var string list
{
    va_list vlst;   
    int i;
   
    char * bufptr;
    char * retptr = s;
   
    va_start(vlst, n);
    for (i=1; i<=n; ++i)   {
       bufptr = va_arg(vlst, char *);
       strcat(s, bufptr);
    }
    va_end(vlst);   
    return retptr;   
}

//********************************************************************************************

#endif
~~~

Chris--A, 2015-08-29:
Nice collection, just watch out for your `fscanf` and `fprintf` when using an Uno, they require half the RAM.

VogonJeltz, 2015-08-29:
sure, I know, but actually I personally just use DUEs.
Additionally, vsscanf does not compile for AVRs at all, and best was if the Arduino IDE would support the real ANSI stdio.h provided by GCC so that no mimics by my ardustdio.h lib had been necessary at all.
Surely also using stdio.h would make one run into memory issues on small cpus, but who hasn't got the RAM is not supposed to #include either lib, that's simply what the ANSI C preprocessor and compiler design is about.

matthijskooijman, 2016-02-04:
@VogonJ, if you want your code to be included in the library manager, you will have to put it in a github repository, add a tag and request addition (see https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ). If you're just looking to share or discuss your code, a github issue is not the right place for this. As mentioned before, issues are for bugs with the code, or (feature) requests (include library manager addition requests). For now, I'll close this issue, feel free to add a comment or create a new issue if you created a library you want included.

closed 2016-02-04 by matthijskooijman

jsotola, 2016-05-24:
just an observation
these two appear to mean the same thing

#define fileIO_OK            +1
#define fileIO_NO_ERR     0

thank you for your work