arduino/Arduino #3743: Arduino IDE suggest new AVR Core for dev builds

NicoHood, 2015-08-28:
I cloned the git master and rebuilt it myself. Now the IDE suggests me a new AVR core.
If you use this core, the HID library will not be available for example.

You have to manually delete the package in the .arduino15/appdata folder.

If you want to let someone test a patch and he clicks update and doesnt know this, this is a problem. Especially because there ARE problems with HID and Windows10 and the new hourly build fixes this.

Also the Firmata lib causes some warnings.

ffissore, 2015-08-31:
This is expected, if the avr core you're using is older than the latest. Ignore the notification if you're not interested in upgrading. Or port your patch to the latest avr core available

NicoHood, 2015-08-31:
The suggested core was **older**, the HID library (pluggable HID) was **missing**.
I built the master branch myself, so it is impossible to be out of date.

ffissore, 2015-08-31:
So you're saying you had core 1.6.8 and the IDE suggested using core 1.6.7?

NicoHood, 2015-08-31:
The IDE says 1.6.7 is installed and wants to download 1.6.8
But 1.6.8 does not contain the HID library. The user in the linked issue above might have the same problem.

I dont know where this 1.6.8 comes from, but it is not up to date with the master (hourly build/custom build).

ffissore, 2015-08-31:
Ah I see. So the IDE is behaving as expected. 1.6.8 has been released with IDE 1.6.5-r5, with genuino boards support. /cc @cmaglie 
@NicoHood please update your copy of master and rebuild

closed 2015-08-31 by ffissore

NicoHood, 2015-09-01:
And why is the HID library not included there (in the newer commit) if my older build includes HID already?

cmaglie, 2015-09-01:
The release of IDE 1.6.5-r5 is a "service" release, it's not meant to provide new functionality except the support for those additional PIDs and drivers for Genuino boards (for this reason the core doesn't contains the HID change you mentioned).

BTW even if we "only" added some pid definitions in the file boards.txt, the presence of the boards manager forced us to make a new release of the AVR core (and to set its version to 1.6.8 because the IDE versions and cores versions are now independent from each other).

This lead to have an AVR core 1.6.8 (basically identical to AVR core 1.6.7), and your hourly build is saying that you have an ""upgrade"" available!

Hope this helps to understand what happened.
This problem should be fixed if you download again the hourly build (or update the git tree and rebuild the IDE).

NicoHood, 2015-09-01:
Isnt that a bit contraproductive? The hourly build is somehow outdated and somehow newer?

The problem is:
I advice people to test (or better USE my PR) and then they upgrade the core (because upgrading is always a good idea for a user) and the PR stops working. And since there is no uninstallating you have to explain them where to delete the file (somwhere in appdata).

Cant we add a flag for hourly builds/PR builds/own compilations to not search for this? Or at least add a dev version which includes the newer hourly fixes and the backported fixes?

Anyways can you please rebuild this PR then to get rid of this issue for my HID Project users?
https://github.com/arduino/Arduino/pull/3697#issuecomment-134308896

cmaglie, 2015-09-01:
> Cant we add a flag for hourly builds/PR builds/own compilations to not search for this? Or at least add a dev version which includes the newer hourly fixes and the backported fixes?

Easier said than done. Those are complex interactions, we can't foresee every possible combination of hourly builds / pull request / production files that may happen.

In this case the Pull Request build from #3697 worked for you because the build included also the HID change (from the main line upstream repository), but the build from PR #3697 is intended to check the .a linkage only.

> Anyways can you please rebuild this PR then to get rid of this issue for my HID Project users?

no problem, I'll trigger another build.

NicoHood, 2015-09-01:
Okay, I know that its easier said than fixed. But know you are aware of the problem, that if you backport fixes some hourly builds/PRs might break on an update.

NicoHood, 2015-10-22:
Please have a look at the linked issue above. This should be fixed. If users want to try a dev build of the IDE and a library that relies on the new dev build it fails to compile. Please ignore the downloaded avr core if the build is newer than the downloaded. Or just mark the current core with a higher version number.

I guess a lot more people will have this error soon as they try to use my library. Or they just give up. And telling the users to delete the avr core inside ./arduino15/ is not that easy for everyone. and even if they do, if they switch back to the stable IDE they get an update notification. This generates a lot of confusion and problems which shouldnt be there.

(to temporary fix this issue as workaround delete ./arduino15/hwardware/avrcore. path might be something like that. For windows in %appdata%)