arduino/Arduino #3746: Arduino IDE compiler should generate full path for .ino file in .elf

maichaell, 2015-08-31:
Arduino IDE compiler generates wrong path information for .ino file in .elf and GDB shows incorrect path. The path to .ino should probably be hardcoded in the generated (.ino to .cpp) .cpp file.

Looking at generated .elf file with GBD shows that path is:  C:\Program Files (x86)\Arduino/Blink.ino

This results in error when debugging the file in Eclipse

[image]

C:\Program Files (x86)\GNU Tools ARM Embedded\4.8 2014q3\bin>arm-none-eabi-gdb d:\Users\me\AppData\Local\Temp\build321087032653580089.tmp\Blink.cpp.elf
GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140731-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>...
Reading symbols from d:\Users\me\AppData\Local\Temp\build321087032653580089.tmp\Blink.cpp.elf...done.
(gdb) info source
No current source file.
(gdb) info sources
Source files for which symbols have been read in:

Source files for which symbols will be read in on demand:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\hooks.c, C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\main.cpp,
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring_digital.c,
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\wiring.c, C:\Program Files (x86)\Arduino/Blink.ino,
D:\Users\me\AppData\Local\Temp\build321087032653580089.tmp\Blink.cpp
(gdb)

maichaell, 2015-09-01:
I get this working with a small workaround. 

Looks like if there is any reference to .ino file in the produced .elf the debugging IDE gets all mixed up. Would it be possible to make this an option in the preferences ?

Here's the workaround:

myblink.ino file contains only this line:

    #include "C:\Program Files (x86)\Arduino\examples\01.Basics\myblink\myblink2.cpp"

and myblink2.cpp contains the actual program.

    #include <arduino.h>
    // the setup function runs once when you press reset or power the board
    void setup() {
      // initialize digital pin 13 as an output.
      pinMode(13, OUTPUT);
      Serial.begin(9600);
    }

    // the loop function runs over and over again forever
    void loop() {
      digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
      delay(100);              // wait for a second
      digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
      delay(100);              // wait for a second
      Serial.println(2);
    }

ffissore, 2015-09-01:
Can you suggest an alternative? Is there a gcc macro/define/pragma to add to the sketch such information? Or maybe a gcc param?

maichaell, 2015-09-01:
That is a good question! I don't know the answer. I will need some more time to investigate this issue and possible solution.

I found out that Compiler.java contains code that writes that "#line" into the .ino. I guess I could try to download Arduino sources and modify that part and try something. I'm thinking to modify it so that it adds full path of .ino into the .cpp. Do you know if that is possible ? 

Maybe that will help but I don't believe it is enough. It might be hard to come up with proper solution without breaking existing functionality. But let's see.

maichaell, 2015-09-02:
processing\app\debug\Compiler.java

    1386: bigCode.append("#line 1 \"" + sc.getFileName() + "\"\n");
    1387: bigCode.append(sc.getProgram());
    1388: bigCode.append('\n');
    1389: bigCount += sc.getLineCount();

In line 1386 we should ensure that sc.getFileName() returns full path. At least for .ino. Now it returns only filename (e.g. Blink.ino) and not the full path. 

How does this sound ?

ffissore, 2015-09-02:
Try hard coding a path that's valid on your pc and try

maichaell, 2015-09-03:
I modifed Arduino IDE so that it does not print #line directives at all into preprocessed .cpp file. I changed some lines in PrePDEProcessor.java and Compiler.java. Only a few small changes.  Then it works okay. It did not help the original problem if I hardcoded full path. 

Can we make this as an advanced option in preferences? 'disable creating #line directives' or something like that. 

If you accept or can consider this kind of option in preferences I am ready to make all the necessary changes and commits to git for that. Thanks.

ffissore, 2015-09-08:
we can't remove line directives: they are used by gcc to report errors at the actual line in the actual file. test it with a sketch made of 2 .ino files and insert an error in the second one (like omittin a `;`)

ffissore, 2015-09-21:
@maichaell have you found a solution which doesn't break existing features?

matthijskooijman, 2015-09-22:
I just modified arduino-builder to output full paths instead of just filenames. The changes are simple, see Arduino/arduino-builder#9. So far, I've found three consequences of doing so:
 - The filename info in the debug symbols is now correct, and gdb can find the source files.
 - Any gcc errors now show a full path, which doesn't look so pretty. This could be fixed by shortening the filename in the console output, perhaps?
 - Any errors are no longer highlighted, I presume that the code that highlights errors only looks for the plain .ino filename. This should be easy to fix.

I haven't done any thorough testing yet, but I'll be running with the change applied for a while.

maichaell, 2015-09-22:
Matthijs, thanks. This is pretty nice fix. Let's hope you can address the other issues too like the error highlighting you mentioned. I haven't tried this fix yet but it is pretty much what I was going after in the beginning.

closed 2015-09-23 by ffissore

ffissore, 2015-09-25:
Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly