arduino/Arduino #3748: Stop compilation on the first error, and highlight it first

cunidev, 2015-08-31:
Highlighting the latest code syntax error makes it more difficult to find the line of code causing problems...

Example:

Class classname();
classname.doThis(); <--- #1
classname.doThat();
classname.doThatAgain();  <--- this line is highlighted in case of error, instead of #1 

I don't know if I'm clear... Thanks

matthijskooijman, 2015-08-31:
I guess the best solution would be to *highlight* all errors, and jump to / focus the first one? Agreed that the last error isn't typically the one to look at and the first error is.

cunidev, 2015-08-31:
Yes, my solution was only if it wasn't possible to highlight all. Of course
all lines highlighted would be better.

mikaelpatel, 2015-08-31:
A simple solution is to use the extra_flags variables in platform.txt and add the gcc compiler option -Wfatal-errors

This option causes the compiler to abort compilation on the first error occurred rather than trying to keep going and printing further error messages. 

The platform variables to change are:
compiler.c.extra_flags=-Wfatal-errors
compiler.cpp.extra_flags=-Wfatal-errors

This could be nice for beginners. 

Ref. 1. https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html

cunidev, 2015-08-31:
Thanks for the answer. Would it be possible to suggest this changes via a
pull request then, or would there be any drawbacks? (There shouldn't be any
in my opinion)

q2dg, 2015-08-31:
+1 for pull request. It's an obvious improvement

matthijskooijman, 2015-08-31:
Having more than one error shown can make things more efficient, though, `-Wfatal-errors` means you can only look at and fix one error at a time (though that might be easiest for novice users, perhaps?).

Perhaps something to discuss on the mailing list?

cunidev, 2015-09-01:
Well, as the code is read by the compiler from the first to the last line, it's not that useful IMHO to see all the errors, because they are usually caused by the first error (like in the first example). Otherwise, it would still be necessary to solve the first error first. Try to remove a simple semicolon: the compiler will probably highlight the last line of the sketch. Pretty useless, isn't it?
I'd insert -Wfatal-errors as default, and in case allow to disable it...

VogonJeltz, 2015-09-01:
I agree to  matthijskooijman:
"
I guess the best solution would be to highlight all errors, and jump to / focus the first one? Agreed that the last error isn't typically the one to look at and the first error is.
"

matthijskooijman, 2015-09-01:
> Well, as the code is read by the compiler from the first to the last line, it's not that useful IMHO to see all the errors, because they are usually caused by the first error (like in the first [...]

That is true, but consider other errors too: a typo in function name, a wrong number of arguments to a function, etc. These will not break the syntax and usually not introduce any additional errors, making any errors beyond the first real errors too.

VogonJeltz, 2015-09-01:
so I'd suggest to implement the list-all-errors-and-at-double-click-then-jump-to-source-code-line feature

Chris--A, 2015-09-01:
Jumping to a line on click seems like a valuable addition. I would probably find having only one error at a time very frustrating.

NicoHood, 2015-09-02:
I also dont like a single error. But what about an option in preferences? Similar to the Warning setting you could stop on the first error here.

Jumping to the first error could be an idea though.

lmihalkovic, 2015-12-04:
is there a sample sketch laying around somewhere that I could use to test the behavior in the way that you'd like it to work? (one of the difficulties is the n -> 1 aggregation of all .INOs but it just means more code...)

lmihalkovic, 2015-12-04:
@mikaelpatel  off topic... where can I find a small self contained sample of a started sketch (the idea is to implements something like "File>New>Cosa Sketch" feature in the IDE - would you like it to be cosablink?)  INCREDIBLE WORK YOU DID!!!

lmihalkovic, 2015-12-04:
I was thinking about 2 panes in the console: 1) for the raw info as it exists now, and the second with a cleaned up list of errors. would that make sense or would you prefer to jump directly from the current raw output list? 

alternatively it is also possible still have a separate list presentation, but available under [QuickAccess](https://github.com/arduino/Arduino/issues/4077) via something like "errors". Meaning from the editor "CRTL+3"  "err" will likely already display the list, then ARROW-UP/DOWN  ENTER  to go somewhere in the source code.

mikaelpatel, 2015-12-04:
@lmihalkovic 
I would actually suggest that a sketch template folder as there is no single start sketch when using Cosa. There are many execution models (events, delayed/timed functions, threads, actors, etc) and several debugging tools (trace, measure, debug, etc). 

Also off the topic: Have you seen the Cosa Debugger? It is a on-target debugger where a set of debug commands may be added to the source code. A debugger shell is entered on break-points, etc. 

The IDE could augment the code with 1) break-point indicator (on line number), 2) variable observe window,  3) memory/stack check, etc. Please have a look at the example sketch. https://github.com/mikaelpatel/Cosa/blob/master/libraries/Debug/examples/CosaDebug/CosaDebug.ino

Cheers!

lmihalkovic, 2015-12-06:
@mikaelpatel  thank you kindly for taking the time to respond...

lmihalkovic, 2015-12-06:
wrote the code, it is indeed helpful (have not settled on the final presentation though... so it is still time to express any views)

looks like the standard builder aborts when one file has any number of issues (am writing a replacement for arduino-builder.. will see what that one will do)

lmihalkovic, 2015-12-29:
This is basically a few more lines of code/cleanup while doing what i suggest in #3771

lmihalkovic, 2016-01-08:
testing bookmarking all errors in the editor (works for my compiler and for arduino-builder - I promise it's not a big deal to change the compiler class to collect all errors instead of bailing on first one - may I suggest that if you go down that path, you consider changing RunningException to become a simple RunningError pojo and no longer an Exception)

[image]

per1234, 2017-07-03:
I don't understand this talk of the last error line being highlighted. I don't remember this ever being the case. The oldest IDE version I have installed is 1.0.3 and even that one jumps to and highlights the first error line. I also checked with 1.6.5-r5 and 1.6.4, which were the recent versions at the time of this report, and still the first error line was jumped to and highlighted.

The idea of jumping to the relevant line when you double click on the error message brought up by shiftleftplusone already has a dedicated issue report: https://github.com/arduino/Arduino/issues/3383

So this issue should be focused solely on the idea of having the option to halt compilation on the first error. I agree this would be very helpful to beginners. Often they don't realize they need to scroll to the top of the console window to see the relevant error message so they are looking at whatever fragment of the end of the error output happens to be visible in the console window.

matthijskooijman, 2017-07-04:
> So this issue should be focused solely on the idea of having the option to halt compilation on the first error.

arduino-builder already stops at the first failed gcc command, so this would need some gcc option to stop at the first error. It seems `-Wfatal-errors` does exactly this. However, I'm not sure if adding this option unconditionally is a good idea. Whereas it might help novice users with simple sketches, it would greatly hinder more advanced users with big sketches and a lot of errors, having to recompile after fixing every one error really slows things down.

tmdrake, 2017-09-05:
There should be an option not to stop on the first error... and jumping to that code. It reallys screws us advance user who may be changing the code while its compiling (and cant cancel)