arduino/Arduino #3750: SPIFFS, limit of 500 characters while adding data to a file.

JeroenBeemster, 2015-08-31:
Hi Igrr,
 Here my 2 issues.

1) I try to use the filesystem SPIFFS (currently only available in staging environment).
 After opening the an existing file with append 'File f = SPIFFS.open(AccessPointsFileName, "a") ' and try to add some text with 'f.println("something") ' everything works fine until I step over a limit of 500 characters (I guess). The worst part of it, is that it will also corrupt the file system and have to reflash it. I solved the reflashing by trial and error. 
 2) How do I "format" the flash memory in the area where the file system is stored.

Here is my test application:

#include "FS.h"
 #define AccessPointsFileName "/accesspoints.txt" 

boolean halt = false;

void setup() {
 Serial.begin(115200);
 if(!SPIFFS.begin()) {
 halt = true;
 Serial.println("Begin failed");

 }

if(!SPIFFS.remove(AccessPointsFileName)) {
 //halt = true;
 Serial.println("remove failed");
 }

Serial.println("Setup done");
 delay(10);
 }

void loop() { 
 if (halt == false) {
 halt = test();
 }
 }

boolean test() {
 Serial.print(".");
 File f = SPIFFS.open(AccessPointsFileName, "a");
 if (!f) {
 Serial.println("file open failed");
 return true;
 }
 Serial.print("Size :");Serial.println(f.size());

f.println("buffer");

f.close();

f = SPIFFS.open(AccessPointsFileName, "r");
 if (!f) {
Serial.println("file open failed");
return true;

}

Serial.print("Size2:");Serial.println(f.size());

f.close();

delay(10);
 return false;
 }

This is the results:

Setup done
 Size :0
 Size2:8
 Size :8
 Size2:16
 Size :16
 Size2:24
 Size :24
 Size2:32
 Size :32
 Size2:40
 Size :40
 .
 .
 Size2:480
 Size :480
 Size2:488
 Size :488
 Size2:496
 Size :496
 file open failed

matthijskooijman, 2015-08-31:
@JeroenBeemster, It seems that your issues are within this SPIFFS library, which (AFAIK) is not distributed with the Arduino IDE. This means that this is not the right bug tracker to submit this bug, so I'm closing this issue (We can reopen if I got this wrong). The right place for this bug is probably with the maintainer or github repository of the SPIFFS library (not sure where this would be, a quick google didn't show me the obvious answer).

closed 2015-08-31 by matthijskooijman

Chris--A, 2015-09-01:
This looks like it is for the ESP8266 (I think your FS.h is [this one](https://github.com/esp8266/Arduino/blob/6d1d13123fb397886b16ca4d49038f544aa62bd0/hardware/esp8266com/esp8266/cores/esp8266/FS.h)).

If this is the case, you can reopen your issue here: https://github.com/esp8266/Arduino