arduino/Arduino #3752: Eclipse-like project file manager on left side

cunidev, 2015-09-01:
I know this would be a big change, but many will agree it would be a positive change too. For the moment I think it would simply be better to have all the project files on the left side instead of the top bar (which is not so nice when working on a large project)... In the future then it would be more easily possible to implement features such as importing external C files (not Arduino-formatted libraries), directories to make projects more tidy... What do you think?

VogonJeltz, 2015-09-01:
I personally don't care where  projects files are listed, nevertheless either extra window should be able to be disabled (e.g., toggle by [F9]: show project file window / hide project file window)

ffissore, 2015-09-01:
This has been requested at least once. I can't find the original issue thought. I guess it was a wontfix, because that will be a major change in both the graphical aspect and way to use the IDE. Arduino Create will feature a side panel

closed 2015-09-01 by ffissore

per1234, 2017-07-09:
I'm not sure whether it's the previous issue @ffissore mentioned but https://github.com/arduino/Arduino/issues/1590 did propose this along with other UI changes.