arduino/Arduino #3755: Use of external editor option

fiechr, 2015-09-01:
Most of the time I'm using another editor like Notepad++ on Windows or Geany on Linux for editing my source files.

The problem is: I'm not really happy with or without enabling the "Use external editor" option.

If it's enabled and I open an example file provided by a library, I can't save it as a new file (which I have to) because the "Save" and "Save as..." options in the File menu are disabled. Also if I just want to open a file and save it under a new name I can't.

If I *don't* use this option, the IDE isn't reloading the file at all (if changed by another program) - even if I reload the file from the open recent list in the File menu! I have to close and reopen the whole IDE - which is a little bit annoying.

There are certainly several "solutions" for this, here are just some ideas:

* Add a "reload file", "refresh" or "reload file from disk" option to the file menu, which allows the user to manually reload a file if he *knows* or assumes it was changed externally.

* Let the IDE always check for changes if the main window lost focus (and regained it).

* Enable the "Save" and "Save as" options even when "Use external editor" option is selected.

* Add a button to actually open an external editor (provided in the preferences) with the current file. This would make the option IMHO much more useful.

* But on the other hand: Is it really necessary to disable editing altogether if an external editor will be used? The idea behind this seems to be to protected the external editor from changes. But **if** someone uses an external editor she/he will most certainly use a potent one which will most certainly be able to detected changes by itself anyway (like Notepad++ does for example).

Somehow I think this option is incomplete and should be reworked. Either by making it easier to use or by removing it altogether and make the IDE more robust regarding external file changes.

VogonJeltz, 2015-09-01:
my suggestion would be is to enhance the Arduino IDE editor by
load (ino, pde, h, c, cpp)
save (current name)
save as (ino, pde, h, c, cpp)
providing all explorer faeturss like e.g., delete/rename/new folder

fiechr, 2015-09-01:
>> Enable the "Save" and "Save as" options even when "Use external editor" option is selected.

> This seems like an easy improvement that would be sufficient for the "external editor" case (though probably only "save as", perhaps)?

Yes, thinking about it "Save as..." is the only option that would make sense to be enabled in this case.

per1234, 2017-07-01:
>- Let the IDE always check for changes if the main window lost focus (and regained it).

This will be done by https://github.com/arduino/Arduino/pull/6074. Would that change solve the issue for you @fiechr? There are test builds available in the pull request if you want to try it out.

per1234, 2020-06-09:
>Add a button to actually open an external editor (provided in the preferences) with the current file. This would make the option IMHO much more useful.

We now have a dedicated feature request for this: https://github.com/arduino/Arduino/issues/8768

per1234, 2021-04-04:
>I can't save it as a new file (which I have to) because the "Save" and "Save as..." options in the File menu are disabled. Also if I just want to open a file and save it under a new name I can't.

Related: https://github.com/arduino/Arduino/issues/5593