arduino/Arduino #3756: Fix 16u2 and 32u4 Bootloader bugs with Genuino

NicoHood, 2015-09-01:
You currently added the Genuino files.
We recently (at least where I was involved) had some issues with the 16u2 firmware of the 16u2 and the 32u4 bootloader. There is some stuff that can be clearly improved.

The question is:
Is this .hex file already in production? Why dont you use this chance to update those firmwares/bootloaders to apply those fixes? I really would like to see this. I am also happy to help improve this, suggest solutions and give more information about what can/should be fixed.

matthijskooijman, 2015-09-01:
@NicoHood, Can you be more specific about these bugs? Link issues or pullrequests for them?

NicoHood, 2015-09-01:
Meh. I wrote them down so many times now.

Mainly the are all fixed in HoodLoader2, so you can keep this as reference:
https://github.com/NicoHood/HoodLoader2/wiki/Overview#hoodloader2-features
https://github.com/NicoHood/HoodLoader2/wiki/Overview#improvementschanges-compared-to-old-leonardo-bootloader

For the 16u2 you could completely use HL2.0.5 if you like to. There is no disadvantage in using HoodLoader2. If you dont use the additional 16u2 reprogramming feature you still have the fixes and 2M full duplex speed.

For 32u4 I'd not use the Serial bridge maybe since this sets the TX/RX pins in the bootloader. I'd keep this optional.

https://github.com/arduino/Arduino/issues/2474
Long discussed. Can be easily then applied to the boards.txt via
https://github.com/arduino/Arduino/pull/3640

https://github.com/arduino/Arduino/pull/3680
https://github.com/arduino/Arduino/pull/3679

Last PR requires a proper reset mechanism for the Leonardo which is currently not provided. The USB Clock is still running after the bootloader, even if you dont want to use the USB functions. Thatswhy you need a stupid workaround otherwise. Not sure what other problems the wrong reset mechanism causes.