arduino/Arduino #3758: [Library Manager] Please add Zumo32U4 library

kevin-pololu, 2015-09-01:
Please add our Zumo32U4 library to the library manager:

https://github.com/pololu/zumo-32u4-arduino-library

Thanks.

ffissore, 2015-09-02:
Added. Thanks!

closed 2015-09-02 by ffissore

PaulStoffregen, 2015-09-02:
@ffissore - Perhaps a guideline should be made, to encourage library publishers to name their Github repositories, their main .h header file, and library consistently?

At least the duplicate library warning that went into 1.6.3 is making conflicts visible to end users, and especially what Arduino is actually doing when a conflict exists.  I recently spent some time helping a couple people resolve some tricky conflicts.  Unfortunately, these are all difficult "corner cases", like when a custom core includes a modified copy of a library, or when someone has created a new library by editing a copy of an existing one, so there isn't any single common use case.

But one universal guideline should be to consistently use the same string for the library name, the main .h header, and directory containing the library.  As a practical matter, that means using the name of the github repository, since users will clone the repository or download the .ZIP archive and keep whatever name came from Github.

For example, in one recent case, a user wrote a sketch using about 10 libraries.  Most were from Adafruit, but a few were enhanced versions, where ambitious users cloned Adafruit's libraries and added extra features which Adafruit didn't wish to support.  They were also using a custom core, which shipped with slightly modified copies of several libraries.

Everything worked well, except for a copy of Adafruit_GFX.h.  Because Adafruit named their repository "Adafruit-GFX-Library" rather than "Adafruit_GFX", and the user left the folder name as it came from Github, the conflict resolution code in the Arduino IDE picked another copy of this library where the folder name exactly matched, even though the user intended to override it.

Until 1.6.3, when users hit these sorts of unexpected library conflicts, they would usually get little help on the forums (they're often unaware of other conflicting copies of a library) and usually they'd end up reinstalling Arduino and starting with a clean slate.  Before 1.6.3, we never really had much opportunity to understand the reasons why users faced library conflicts.

I'm still learning some of the strange issues that really happen with Arduino users who have lots of copies of the Arduino IDE (including forks like Energia) and lots of third party cores and tons of libraries, including forks and customized copies.  It can get really complicated.

But one simple problem is at the heart of nearly all user confusion.  Libraries that lack perfect consistency in the library name, primary .h header file, folder name and github repository name end up causing tremendous confusion for users.

DavidEGrayson, 2015-09-02:
If you use GitHub to install the Adafruit GFX Library (and ignore Adafruit's instructions), the folder name will be "Adafruit-GFX-Library-master".  If you use the Library Manager to install it, the folder name will be "Adafruit_GFX_Library".  Either way, this is not an exact match to "Adafruit_GFX".

To make the library manager behave properly, I think Adafruit could just change the "name" property (which is in [library_index.json](http://downloads.arduino.cc/libraries/library_index.json) and [library.properties](https://github.com/adafruit/Adafruit-GFX-Library/blob/0acee33cfe8b500c5a2a1c4f29913746936b53ba/library.properties#L1)) from "Adafruit GFX Library" to "Adafruit_GFX".  The repository doesn't need to be renamed, but maybe someone should encourage library authors to choose correct names in library.properties.

But what about users installing from GitHub?  Since GitHub adds a version suffix like "-master" to the directory name, exact matches between header file names and folder names will not be a reliable way of resolving library conflicts.  Maybe it would be better if the conflict resolver used some information from library.properties, such as the "name" property, or perhaps a list of header files.  Making that change would mean that library authors no longer have to tell users to rename directories, and users who don't rename the directories would still be able to use the library properly.  (As a side effect, extra information can then be stored in the directory name, such as the version of the library or the person who produced it, which might help people organize their libraries and understand everything that they have installed.)

PaulStoffregen, 2015-09-02:
> but maybe someone should encourage library authors to choose correct names in library.properties.

Yes, this!

> But what about users installing from GitHub? Since GitHub adds a version suffix like "-master" to the directory name,

The Arduino IDE has code which specifically looks for the "-master" suffix.