arduino/Arduino #3762: Add finite state machine library to Library Manager

jonblack, 2015-09-03:
The URL for the github repository is: https://github.com/jonblack/arduino-fsm

PaoloP74, 2015-09-03:
--> https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification
--> https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x---package_index.json-format-specification

jonblack, 2015-09-03:
Is there a mistake in the `library.properties` file or do I just need to add a `package_index.json`?

PaoloP74, 2015-09-03:
Add the .json file and make the subdirectory "src".

jonblack, 2015-09-03:
According to the first link, the `src` folder is for 1.5.x-only libraries.

> For 1.5.x-only libraries, the source code resides in the src folder.

I'd like my library to be for all versions. Has this changed? If it has, another library (which has been accepted) of mine also needs updating.

As for the package_index.json, isn't that so the Boards Manager of the Arduino IDE can be used to automatically install support for 3rd party hardware? If so, my library doesn't need it.

PaoloP74, 2015-09-03:
@cmaglie

cmaglie, 2015-09-04:
Library added! thanks.

Just to clarify a bit:

- A library is 1.5 compliant even if the old layout is used (i.e. without the `src` folder). The advantage is that it works for 1.0.x IDE, the drawback is that you can't use the recursive sub-folders compile.

- The document https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x---package_index.json-format-specification is about "cores" and not "libraries", so it's useful if you're going to produce a new board and you want to leverage the "Boards manager".
In this case it doesn't contains relevant information about the `library.properties` file.

closed 2015-09-04 by cmaglie