arduino/Arduino #3765: serial communication lib for 2-way half duplex communication by 64-128byte bandwidth

VogonJeltz, 2015-09-04:
hello,
I just started with implementing a serial communication lib between 2 Arduino Megas (Serial2) for 2-way half duplex communication by 64-128byte bandwidth (sending amd receiving data packages byte arrays[128] ) . 
Baud rate is 460800. 
I started by byte arrays[8], enlarging the array size step by step.
But unfortunately I encountered many transmission errors, data corruption, and data loss even if I tried to implement sort of basic data sync and error check, admittedly I'm jsut at my very at the beginning of my experiments.

So before I try to invent the wheel from the scratch:
is there already an existing comm lib for data array packages available?

PaulStoffregen, 2015-09-04:
This github issue tracker is for reporting bugs in Arduino.

You really need to post this question (and most of the stuff you post here) on the Arduino Forum:

https://forum.arduino.cc/index.php?board=3.0

VogonJeltz, 2015-09-04:
ok,
the issue is: I can't find it, 

and so if it really doesn't,
then my request is to pull this for an improvement!

PaulStoffregen, 2015-09-04:
Post your question on the forum.

VogonJeltz, 2015-09-04:
no.

VogonJeltz, 2015-09-05:
I searched all forums, the playground and all existing code examples in the IDE  but no code examples found!

Please provide code examples  about quick and fail-safe 2-way data transmission of long strings .or byte arrays (length > 48 bytes, possibly up to 128 bytes each.)

closed 2015-09-07 by ffissore