arduino/Arduino #3766: Documentation error for SoftwareSerial - inverse_logic parameter not mentioned

nickgammon, 2015-09-05:
On the documentation for the SoftwareSerial constructor:

https://www.arduino.cc/en/Reference/SoftwareSerialConstructor

> SoftwareSerial(rxPin, txPin)

There is no mention of the **inverse_logic** argument. See the source:

```C++
public:
  // public methods
  SoftwareSerial(uint8_t receivePin, uint8_t transmitPin, bool inverse_logic = false);
```

This is confusing for people who read code which uses that third argument. It should be mentioned on the page as an optional argument with an explanation along the lines of:

> **inverse_logic** is used to invert the sense of incoming bits (the default is normal logic).
>
> If set, SoftwareSerial treats a LOW (0 volts on the pin, normally) on the Rx pin as a **1**-bit (the idle state) and a HIGH (5 volts on the pin, normally) as a **0**-bit. It also affects the way that it writes to the Tx pin.
> **Warning**: You should not connect devices which output serial data outside the range that the Arduino can handle, normally 0V to 5V, for a board running at 5V, and 0V to 3.3V for a board running at 3.3V.

agdl, 2015-09-07:
@nickgammon added as you suggested. If it is ok for you close the issue please, otherwise other suggestions are good :)

nickgammon, 2015-09-07:
The heading on the page still reads:

> SoftwareSerial(rxPin, txPin)

agdl, 2015-09-07:
done

nickgammon, 2015-09-08:
Good. :)
One more thing: On this line:

> A call to SoftwareSerial(rxPin, txPin, inverse_logic) creates a new SoftwareSerial object, whose name you need to provide as in the example below. 

You should probably mention that inverse_logic is an optional parameter. eg.

> A call to SoftwareSerial(rxPin, txPin, inverse_logic) creates a new SoftwareSerial object, whose name you need to provide as in the example below. The **inverse_logic** argument is optional and defaults to false. See below for more details about what it does.

In fact, now that I look at it more closely, the example is not very good. It calls the constructor twice.

```C++
#define rxPin 2
#define txPin 3

// set up a new serial port
SoftwareSerial mySerial =  SoftwareSerial(rxPin, txPin);
```

That should read:

```C++
#include <SoftwareSerial.h>

const byte rxPin = 2;
const byte txPin = 3;

// set up a new serial object
SoftwareSerial mySerial (rxPin, txPin);
```

No need to construct a temporary object and then immediately get rid of it.

So let's reword to:

> SoftwareSerial is used to create an instance of a SoftwareSerial object, whose name you need to provide as in the example below. The **inverse_logic** argument is optional and defaults to false. See below for more details about what it does. Multiple SoftwareSerial objects may be created, however only one can be active at a given moment.

agdl, 2015-09-08:
@nickgammon actualkly I stated it in the inverse_logic parameter description, but its better to repeat that is optional! Done if it looks good close the issue please!

nickgammon, 2015-09-08:
Looks good!

closed 2015-09-08 by nickgammon

andyb200489, 2017-12-13:
As a newbie I found the example very confusing and it was trial and error to make my inverted logic display work. The line should show how many parameters can be passed to the function and the form they take. As a newbie I dis not know if I should use True or 1.