arduino/Arduino #3767: Problems in "Serial Plotter" Y-axis resolution (1.6.6 Hourly build 2015/09/04)

q2dg, 2015-09-06:
Hello.
I've detected two problems. 

**First**
If I execute...:

    void setup() {  Serial.begin(9600); }
    void loop() {  Serial.println(2);  }

...value 2 is shown at the top of Serial Plotter but Y-axis doesn't show the numerical coordinates. If I put a greater value (like 10, for instance), then it's fine.

**Second**
If I execute...

    void setup() {  Serial.begin(9600); }
    void loop() {  Serial.println(random(55));  }

...graph is shown fine, but if I add a delay after Serial.println() **(only if delay time is more than 700ms aprox.!!)** , the Y-axis boundaries are not updated accordingly, showing always from 0 to 400 values (and not from 0 to 60 or similar, as expected).

Thanks

ffissore, 2015-09-08:
Fixed. Fix will be available in next hourly build. http://www.arduino.cc/en/Main/Software#hourly

As for the effect due to the delay, it's due to the size of the buffer used by the plotter: Y axis uses min and max buffer values, some initial value is higher than all the subsequent values, but until the buffer exhausts (and the plot slides to the left) that Y axis will keep on show greater values than expected

closed 2015-09-08 by ffissore

q2dg, 2015-09-10:
Ok, thanks

commit 4cb72ceb (2015-09-08): Code cleanup
SerialPlotter now uses min/2 and max*2 instead of plain min and max
Ticks works also when min == max
Fixes #3767