arduino/Arduino #3771: Compile/Upload must retain control over the running processes

ffissore, 2015-09-08:
Right now processes are started but they are not under user control. User cannot stop or abort them, and hitting the buttons twice starts twice the processes.
Symptoms have been acknowledged in #3568 #2718 #2574 #2417 #766

lmihalkovic, 2015-12-04:
[LMDONE]

lmihalkovic, 2015-12-29:
It was straight forward to make the compilation process interruptible cleanly anytime with a java written driver. Dealing with arduino-builder cleanly is not as simple. Someone will have to modify builder to support this feature.

facchinm, 2016-02-05:
Solved by #4482 (commit 3f07b8a)

facchinm, 2016-04-11:
Fixed by merging 6d5597b070c7bedce15047c7c560249965cc0bfc

NicoHood, 2016-04-11:
This is not fixed:
https://github.com/arduino/Arduino/issues/3568

Upload a (larger) program to a 32u4 and unplug the device when uploading the code. It will freeze. And you still can reconnect, select a new port and upload again. So a) two processes are started and b) you still cannot abort.

facchinm, 2016-04-11:
The second time you click the upload button the stuck `avrdude_bin` gets killed (just tested with nighly, a Leonardo and an ArduBoy sketch). Are you on Linux ?  (I believe so but I'm asking just to be sure :smile: )

NicoHood, 2016-04-11:
Sure I am on Linux ;)
Me as user (and as developer ;( ) expect an "abort" button. I am very afraid of pressing uploading twice, and after arduino was buggy about that for a long time now, me and others are afraid to even try this again. Its also not expected behavior. So a real "kill" aka abort button is appreciated.

NicoHood, 2016-04-11:
Testato we got this super thumbs up github feature now. Please use it.

Testato, 2016-04-11:
I do not understand :-)

NicoHood, 2016-04-11:
I am not sure if this is ironie. On the upper right hit the smiley button and select a thumbs up instead of posting +1

Testato, 2016-04-11:
now i understand, on the smartphone the new icon systems it is not visible.
So if you post by a smartphone, or directly by email, it is not possible use it

facchinm, 2016-04-12:
@NicoHood , I agree with you, an abort button could be far more understandable (it should only appear during upload, since compilation is going to end sooner or later).
I'm asking the UI team if it is the desirable feedback (then the PR should come in a matter of minutes, all the pieces and callbacks are in place )

matthijskooijman, 2016-04-12:
@facchinm, I'm also planning to revisit this part of the code as part of my debugger work, so perhaps we should have a chat, probably including the UI folks as well?

matthijskooijman, 2016-04-12:
woops, didn't mean to reopen

closed 2016-04-12 by matthijskooijman