arduino/Arduino #3772: Dynamixel_Servo library

michaelkrzyzaniak, 2015-09-09:
https://github.com/michaelkrzyzaniak/Dynamixel_Servo

ffissore, 2015-09-09:
So what?

michaelkrzyzaniak, 2015-09-09:
Hi. According to https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ, in order to add a library to the Arduino library manager I should, amongst other things, "Open an issue on Arduino's github, specifying the git repo (or github url) from where to download your library." I hope I have not misunderstood these instructions.

Chris--A, 2015-09-10:
I think you'll need to move the contents of the directory *Dynamixel_Servo* into the top level of the repo. I do not think the library manager will look in sub folders for the **library.properties** file.

ffissore, 2015-09-10:
@michaelkrzyzaniak I suspected you were asking for inclusion, but I expected some more words. See #3769 for example. You're email looked like spam at a first glance. I'm adding it

ffissore, 2015-09-10:
Added. Thanks!

closed 2015-09-10 by ffissore