arduino/Arduino #3777: compiler optimization bug? (or: what do I miss?)

VogonJeltz, 2015-09-10:
hi,
I just came upon this issue by incident -
in the following code the LED does not blink  :o
what do I miss ?
first I thought is was a compiler optimization thing, but the program works faulty even after I made the variables "volatile"  
(IDE 1.6.5, Arduino Mega)

~~~

void setup(){
   pinMode(13, OUTPUT);  
}

void loop(){
     volatile uint32_t  i, b;   
     
     b=0;
     digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
     for( i = 0; i < 2147483647 ; i++){   b++;}             // wait.. a little longer...?
     digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
     b=0;
     for( i = 0; i < 2147483647 ; i++){   b++;}             // wait
}

~~~

it's just for experimental reasons, and so that issue is actually the issue  
Because the point is:
the compiler must not care about out-optimizing variables if they are "volatile" !
So actually the code had to be compiled without out-optimizing the loops, and therefore I expected the LED to blink!

cmaglie, 2015-09-10:
@VogonJeltz 
This issue-tracker is for tracking bugs on the Arduino platform not to give advice on generic user's projects.
The forum is a much appropriate place to ask for help: http://forum.arduino.cc/

Please post here only when you are sure that there is a bug to fix.

Thank you.

closed 2015-09-10 by cmaglie

Arjan-Woltjer, 2015-09-10:
Question, how long did you wait for the led to start blinking? More or less than 2000 seconds?

VogonJeltz, 2015-09-10:
yes, it was a mental arithmetic issue 8-)