arduino/Arduino #3778: Cancelling index file download in Library Manager fails and crashes

matthijskooijman, 2015-09-11:
When the library manager is opened, it downloads new index files. Since I'm on a slow connection, I clicked cancel, resulting in this:

[image]

After a few seconds, it continued downloading, but with a different total size, so I presume that it downloads a second file (the feedback about this could perhaps also be improved, now it's surprising that it "starts over" at 0kb halfway). After clicking cancel for a second time, I got:

[image]

Here is the full backtrace from the log window:

```
 Unexpected end-of-input in field name
 at [Source: java.io.FileInputStream@7861404e; line: 9738, column: 92] (through reference chain: LibrariesIndex["libraries"])
java.lang.RuntimeException: com.fasterxml.jackson.databind.JsonMappingException: Unexpected end-of-input in field name
 at [Source: java.io.FileInputStream@7861404e; line: 9738, column: 92] (through reference chain: LibrariesIndex["libraries"])
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onUpdatePressed$13(LibraryManagerUI.java:211)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI$$Lambda$81/484702524.run(Unknown Source)
	at java.lang.Thread.run(Thread.java:745)
Caused by: com.fasterxml.jackson.databind.JsonMappingException: Unexpected end-of-input in field name
 at [Source: java.io.FileInputStream@7861404e; line: 9738, column: 92] (through reference chain: LibrariesIndex["libraries"])
	at com.fasterxml.jackson.databind.JsonMappingException.wrapWithPath(JsonMappingException.java:232)
	at com.fasterxml.jackson.databind.JsonMappingException.wrapWithPath(JsonMappingException.java:197)
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.wrapAndThrow(BeanDeserializerBase.java:1311)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeFromObject(BeanDeserializer.java:297)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:121)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:2888)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2080)
	at cc.arduino.contributions.libraries.LibrariesIndexer.parseIndex(LibrariesIndexer.java:91)
	at cc.arduino.contributions.libraries.LibrariesIndexer.parseIndex(LibrariesIndexer.java:78)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:657)
	at processing.app.Base$10.onIndexesUpdated(Base.java:1238)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onUpdatePressed$13(LibraryManagerUI.java:209)
	... 2 more
Caused by: com.fasterxml.jackson.core.JsonParseException: Unexpected end-of-input in field name
 at [Source: java.io.FileInputStream@7861404e; line: 9738, column: 92]
	at com.fasterxml.jackson.core.JsonParser._constructError(JsonParser.java:1369)
	at com.fasterxml.jackson.core.base.ParserMinimalBase._reportError(ParserMinimalBase.java:532)
	at com.fasterxml.jackson.core.base.ParserMinimalBase._reportInvalidEOF(ParserMinimalBase.java:465)
	at com.fasterxml.jackson.core.json.UTF8StreamJsonParser.parseEscapedFieldName(UTF8StreamJsonParser.java:1686)
	at com.fasterxml.jackson.core.json.UTF8StreamJsonParser.parseLongFieldName(UTF8StreamJsonParser.java:1522)
	at com.fasterxml.jackson.core.json.UTF8StreamJsonParser.parseMediumFieldName(UTF8StreamJsonParser.java:1506)
	at com.fasterxml.jackson.core.json.UTF8StreamJsonParser._parseFieldName(UTF8StreamJsonParser.java:1441)
	at com.fasterxml.jackson.core.json.UTF8StreamJsonParser.nextToken(UTF8StreamJsonParser.java:668)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeFromObject(BeanDeserializer.java:288)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:121)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:230)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:207)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:23)
	at com.fasterxml.jackson.databind.deser.SettableBeanProperty.deserialize(SettableBeanProperty.java:464)
	at com.fasterxml.jackson.databind.deser.impl.MethodProperty.deserializeAndSet(MethodProperty.java:98)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserializeFromObject(BeanDeserializer.java:295)
	... 10 more
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at cc.arduino.contributions.libraries.LibrariesIndexer.rescanLibraries(LibrariesIndexer.java:111)
	at cc.arduino.contributions.libraries.LibrariesIndexer.setLibrariesFolders(LibrariesIndexer.java:104)
	at processing.app.BaseNoGui.onBoardOrPortChange(BaseNoGui.java:814)
	at processing.app.Base.onBoardOrPortChange(Base.java:1222)
	at processing.app.Base.openLibraryManager(Base.java:1254)
	at processing.app.Base.lambda$rebuildImportMenu$26(Base.java:1142)
	at processing.app.Base$$Lambda$39/1882349076.actionPerformed(Unknown Source)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.menuDragMouseReleased(BasicMenuItemUI.java:943)
	at javax.swing.JMenuItem.fireMenuDragMouseReleased(JMenuItem.java:586)
	at javax.swing.JMenuItem.processMenuDragMouseEvent(JMenuItem.java:483)
	at javax.swing.JMenuItem.processMouseEvent(JMenuItem.java:429)
	at javax.swing.MenuSelectionManager.processMouseEvent(MenuSelectionManager.java:329)
	at javax.swing.plaf.basic.BasicPopupMenuUI$MouseGrabber.eventDispatched(BasicPopupMenuUI.java:870)
	at java.awt.Toolkit$SelectiveAWTEventListener.eventDispatched(Toolkit.java:2425)
	at java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2317)
	at java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2316)
	at java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2316)
	at java.awt.Toolkit.notifyAWTEventListeners(Toolkit.java:2275)
	at java.awt.Component.dispatchEventImpl(Component.java:4777)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```

ffissore, 2015-09-25:
Fixed. Fix will be available with next hourly build http://www.arduino.cc/en/Main/Software#hourly

closed 2015-09-25 by ffissore

matthijskooijman, 2015-09-25:
Awesome! The crash is now gone, though there is still a short moment where the "close" button is shown, but the "cancel" button is not yet hidden (as in the first screenshot). Just a minor thing, though.

uomodellamansarda, 2016-11-02:
@ffissore I got the same problem, how can i install the fix? 
Is my first time withi Genuino, i have downloaded the file but i don't know how to continue. 
Where i have to extract the files? 
I have to open arduino.exe?

ffissore, 2016-11-02:
@uomodellamansarda sorry you missed the news. I no longer work for arduino. See https://groups.google.com/a/arduino.cc/forum/#!msg/developers/YJLX6AZHem4/kpsLef7XBAAJ

uomodellamansarda, 2016-11-02:
@ffissore tnx!

commit f511c245 (2015-09-25): Fallback to stock library_index.json file if the copy in .arduino15 is missing/corrupted or if it the downloaded one if corrupted. Fixes #3778